dsmrreader/dsmr-reader #1324: Netto verbruik

cmderks93, 2021-03-18:
Wij hebben 3 fase, met ook 3 fase zonnepanelen.
Hierdoor is het dus geregeld zo dat wij verbruik en teruglevering tegelijk hebben.
De meter maakt hier netjes een netto verbruik van, waardoor er altijd maar 1 telwerk loopt.

Helaas zie ik dit niet terug in de telegrammen, en ook niet in deze software. Is het een idee om er een "netto verbruik" bij in te zetten?

dennissiemensma, 2021-03-18:
Bedankt voor je vraag. Het is mij niet helemaal duidelijk wat je bedoelt. DSMR-reader weergeeft de gegevens zoals ze uit de slimme meter komen. Heb je een voorbeeld en/of screenshot van waar je op doelt dat er mis gaat of mist?

cmderks93, 2021-03-18:
Zal proberen het duidelijker uit te leggen. (Screenshot maak ik morgen)
(Dit geld trouwens alleen voor de 3 fase aansluitingen)

Veronderstel dat ik 3000w opwek met de zonnepanelen. Dat is 1000w per fase.
Op fase 1 heb ik de wasmachine aan staan, deze vraagt 2000w. Op fase 2 en 3 heb ik geen verbruikers aan staan. 
Hierdoor zal het er in de telegram als volgt uitzien.
Afname fase 1 1000w (want 2000w verbruik - 1000w opwek is netto 1000w afname van het net)
Afname fase 2 0w
Afname fase 3 0w
Teruglevering fase 1 0w
Teruglevering fase 2 1000w
Teruglevering fase 2 1000w
Totaal afname 1000w
Totaal teruglevering 2000w

Dsmr-reader zal aangeven: teruglevering 2000w
De 1000w afname zie ik dan niet terug.
Maar op de meter zelf, lees ik een verbruik af van 1000w 
Dit omdat de meter intern over de fases saldeerd.

Helaas is dit alles al een gebrek in de dsmr, want deze waarde zie ik nergens terug in de telegrammen.

Het zou mooi zijn als dsmr-reader dus ook een netto afname/teruglevering krijgt. Dit kan door middel van de beide totalen (afname en teruglevering) van elkaar af te trekken.

dennissiemensma, 2021-03-18:
Bedankt voor je aanvulling. Als het goed is kun je deze Watt-gegevens al terugzien in de grafiek "_Recent distribution of electricity phases_", per fase per richting. 
Ik kan je situatie hier niet nabootsen, maar ik kan me voorstellen dat als je zowel 3 fases import als export hebt, de slimme meter die gegevens wel invult op de manier die je verwacht.

Je kunt kijken of je deze velden terug kan vinden in je telegram:
```
            1-0:21.7.0(01.111*kW)
            1-0:41.7.0(02.222*kW)
            1-0:61.7.0(03.333*kW)

            1-0:22.7.0(04.444*kW)
            1-0:42.7.0(05.555*kW)
            1-0:62.7.0(06.666*kW)
```

Niet te verwarren met deze, wat de saldering is vermoed ik en wat ook op het dashboard getoond wordt:
```
            1-0:1.7.0(01.193*kW)
            1-0:2.7.0(00.000*kW)
```

Alleen toont het dashboard nu wel of het ene of het andere en niet beiden. Wellicht zorgt dat ook nog voor verwarring bij je.

dennissiemensma, 2021-03-18:
En dat laatste is dan nog wel aan te passen in DSMR-reader, maar dan heb ik wel een telegram van je nodig in die exacte situatie om te zien wat er gebeurt.

cmderks93, 2021-03-18:
Volgens mij heb je het helemaal begrepen!
"Recent distribution of electricity phases" hier staat alles inderdaad in.

Niet de ruwe telegram kunnen vinden. Wel de laatste, maar daar heb ik nu het donker is niet zoveel aan.
Morgen zal ik even de ruwe telegram opvangen en doorsturen. (Als de zon het toelaat)

Electricity currently delivered:
0,202
Electricity currently returned:
0,223

Phase currently delivered l1:
0,000
Phase currently delivered l2:
0,202
Phase currently delivered l3:
0,000

Phase currently returned l1:
0,107
Phase currently returned l2:
0,000
Phase currently returned l3:
0,116

dennissiemensma, 2021-03-18:
Je kunt het telegram overigens ook hier vinden: `/admin/dsmr_datalogger/meterstatistics/`

Haal wel even de serienummers weg van je meter(s).

cmderks93, 2021-03-19:
Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(xxxxx)
0-0:96.1.1(xxxxc)
1-0:1.8.1(003414.206*kWh)
1-0:1.8.2(005993.165*kWh)
1-0:2.8.1(000181.633*kWh)
1-0:2.8.2(000385.323*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(01.996*kW)
1-0:2.7.0(00.252*kW)
0-0:96.7.21(00622)
0-0:96.7.9(00005)
1-0:99.97.0(1)(0-0:96.7.19)(xxxxx)(0000000189*s)
1-0:32.32.0(00002)
1-0:52.32.0(00002)
1-0:72.32.0(00002)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(236.0*V)
1-0:52.7.0(239.0*V)
1-0:72.7.0(238.0*V)
1-0:31.7.0(008*A)
1-0:51.7.0(001*A)
1-0:71.7.0(001*A)
1-0:21.7.0(01.983*kW)
1-0:41.7.0(00.013*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.252*kW)
0-1:24.1.0(003)
0-1:96.1.0(xxxxx)
0-1:24.2.1(xxxxx)(03155.516*m3)
!E84C

dennissiemensma, 2021-03-19:
Ik heb het aangepast voor de volgende versie. Ze zijn dan beiden tegelijkertijd zichtbaar.

[image]

closed 2021-03-19 by dennissiemensma

cmderks93, 2021-03-20:
Super!

Is het misschien een idee hier ook (optioneel) een totaal van te kunnen maken? Zodat je exact hetzelfde ziet als fysiek op de meter. In het voorbeeld dus 1384w

dennissiemensma, 2021-03-20:
Ik kan me vinden in je suggestie, maar ik denk dat dat eerlijk gezegd wat overkill wordt.

cmderks93, 2021-03-21:
Persoonlijk zou ik dit zeker niet overkill vinden. Het is immers een gegeven die de meter ook hanteert maar helaas niet in de telegrammen zet.
Desalniettemin is de keuze gerespecteerd.

Daarnaast heb ik wel een andere opmerking:
Als er teruglevering en afname tegelijk plaatsvind, klopt de tarief weergave hierbij niet. Zo had ik vanmiddag een teruglevering van 177w welke vervolgens 39ct per uur zou kosten.

dennissiemensma, 2021-03-21:
Bedankt voor je aanvulling. Dat laatste is inderdaad ook aangepast, met bovenstaande wijziging, en hing samen met de eerdere aanname dat er altijd maar 1 kant op geleverd wordt.

dennissiemensma, 2021-03-23:
Uitgebracht in [v4.14](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.14.0)

dennissiemensma, 2021-03-24:
@cmderks93 ongerelateerd, maar draai je toevallig een handmatige installatie van DSMR-reader of gebruik je Docker?

cmderks93, 2021-03-24:
Ik draai gewoon op Ubuntu, met de datalogger op een Pi
Gewoon jou snelle installatie methode gebruikt.
Hoezo?

dennissiemensma, 2021-03-25:
Ik ben wat aan het experimenteren met de gemixte weergave van verbruik/teruglevering, alleen kan ik dat zelf niet testen met natuurlijke data, dus ik zoek wat testgebruikers met zonnepanelen voor hun ervaring. 
In dit geval specifiek om te kijken of een "rainfall" grafiek-variant uitkomst biedt of dat het juist voor meer verwarring zorgt. Voorbeeld: https://echarts.apache.org/examples/en/editor.html?c=area-rainfall

Lijkt je dat wat om later deze week te proberen? Je kunt op elk moment terug naar de huidige versie.

cmderks93, 2021-03-25:
Dat lijkt me wel wat.
Ik ben ook wat aan het experimenteren met een 2e installatie welke niet lokaal draait. Ik kan je eventueel daar wel toegang toe geven, of voor een periode onze telegrammen via de API aanleveren aan je.

dennissiemensma, 2021-03-25:
Bedankt voor je aanbod! Wellicht dat ik er op een later moment gebruik van maak. Voor nu is het denk ik het handigste dat je eerst zelf even kijkt of de opzet in je eigen situatie uberhaupt werkbaar is. 
Het kan namelijk zijn dat de (twee) rainfall grafieken juist minder duidelijk zijn omdat jij in jouw geval zowel verbruik als teruglevering tegelijkertijd kan hebben.

Meer info en een how-to vind je hier: https://github.com/dsmrreader/dsmr-reader/issues/1000#issuecomment-805963572

dennissiemensma, 2021-03-25:
_Zorg er trouwens wel voor dat je eerst op 4.14 draait. Want de branch is gebaseerd op die release en anders krijg je zowel 4.14 als de beta-variant in tegelijkertijd (kan geen kwaad, maar is wat minder transparant wellicht)_

AbelLykens, 2021-03-31:
Een negatief Return-getal lijkt op verbruik, of is het alleen mijn hoofd die zo denkt? ;)

dennissiemensma, 2021-03-31:
@AbelLykens als het goed is is dat voor de volgende 4.15 release aangepast

AbelLykens, 2021-03-31:
Held.

AbelLykens, 2021-04-05:
Nice helemaal top nu. Ook die negatieve grafieken zijn fijn!

pa1pdr, 2021-06-10:
Ik ben niet zo gecharmeerd van die negatieve grafiek op teruglevering, temeer om dat de y-as nu verschillende schalen heeft. Daarnaast neemt die negatieve teruglevering veel screen-estate weg. Kan het een optie worden?

dennissiemensma, 2021-06-10:
@pa1pdr bedankt voor je aanvulling. Voor het schalen is nog een issue wat eerst opgepakt moet worden. Afhankelijk daarvan is denk ik beter te zeggen of het uiteindelijk werkbaar is of niet zo onduidelijk.

rodeho, 2021-08-07:
> Super!
> 
> Is het misschien een idee hier ook (optioneel) een totaal van te kunnen maken? Zodat je exact hetzelfde ziet als fysiek op de meter. In het voorbeeld dus 1384w

Ja heel graag een schakelaar voor de oude implementatie. Mijn externe databronnen zijn nu niet in staat de juiste data meer te laten zien. 

https://github.com/dsmrreader/dsmr-reader/issues/1431#issuecomment-894646771

dennissiemensma, 2021-08-07:
> Ja heel graag een schakelaar voor de oude implementatie.

De oude implementatie was sowieso niet correct, daar miste dus het verbruik uberhaupt in sommige situaties en dat was een bug. Overigens puur een visuele bug.

> Mijn externe databronnen zijn nu niet in staat de juiste data meer te laten zien.

De API is niet veranderd met deze wijziging, dat was puur visueel in het dashboard.

rodeho, 2021-08-07:
> > Mijn externe databronnen zijn nu niet in staat de juiste data meer te laten zien.
> 
> De API is niet veranderd met deze wijziging, dat was puur visueel in het dashboard.

Dennis ik heb helaas niet de kennis & kunde omdat zelf te controleren/aan te passen. De website welke ik gebruik is de volgende: https://github.com/amrij/zonnepanelen/blob/master/live-server-data-electra-dsmr.php ik denk dat deze file de data ophaalt.

Ik sluit toch niet uit dat de API, onbedoeld, onderhevig is geweest aan een change, aan jouw zijde. De tijdslijn valt zo dicht op de 4.14 verandering. Zou je daar iets over kunnen roepen/controleren als je o.a. deze snippet ziet? Hierin wordt de live data opgevraagd?

Als ik het codeblok juist heb dan krijg je op de website nu slechts het verbruik van de 2 fases waarin de zonnepanelen niet zitten. Het getal 108watt zoals uit de som uit mijn eigen topic. Misschien handig om daar verder te gaan?

```
// Get current info for P1_ElectriciteitsMeter from DSMR server
if ($period == 'c' ) {
	//Get current info for P1_ElectriciteitsMeter from DSMR server
	$response = file_get_contents($dsmr_url.'/api/v2/consumption/electricity-live', null, stream_context_create(array(
			'http' => array(
			'method' => 'GET',
			'header' => array('X-AUTHKEY:'.$dsmr_apikey))
			)
	));

```