dsmrreader/dsmr-reader #1325: teveel onverwerkte telegrammen

ronniebee, 2021-03-19:
```

Ben al uren bezig mijn reader weer aan de praat te krijgen. 
configuratie""
Docker
>home assistant 

Heb de usb poort aangepast ( was veranderd van usb1 naar usb0)
draai de laatste versie van xirzix (docker compose)

wie kan mij adviseren?

DSMR-READER
    App / Python / Database                                                       v4.13 / v3.9.2 / postgresql
    BE sleep / DL sleep / Retention / Override                                     5.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              25631
    Consumption records electricity / gas (est.)                                                   1396 / 280

UNRESOLVED ISSUES
    Te veel onverwerkte telegrammen: 11703                                                     19 uur geleden
    Dagstatistieken lopen achter                                                        1 dag, 14 uur geleden

```

dennissiemensma, 2021-03-19:
Bedankt voor je melding. Ik denk dat je het beste kunt beginnen door in de logfiles te kijken. Zie je daar toevallig foutmeldingen?

ronniebee, 2021-03-19:
Hoi Dennis, ik kan deze files niet benaderen.  als ik :

docker$ sudo supervisorctl status
[sudo] wachtwoord voor XXXX: 
sudo: supervisorctl: opdracht niet gevonden

Ben geen expert met terminal :-)

dennissiemensma, 2021-03-19:
Ik kan je op Docker-gebied niet echt helpen. Je kunt het beste zelf even oude issues zoeken voor logs, wellicht werkt dit voor je: xirixiz/dsmr-reader-docker/issues/158

closed 2021-03-19 by dennissiemensma

dennissiemensma, 2021-03-20:
Bleek veroorzaakt door #1282