dsmrreader/dsmr-reader #1326: Error in upgrade van 4.12 naar 4.13 tijdens de-installatie van six

reboot1983, 2021-03-20:
Hi,

Tijdens de upgrade van .12 naar .13 kreeg ik onderstaande foutmelding in de console. Ik ben niet helemaal zeker of het 'critial' is, want na de upgrade lijkt dsmr-reader wel gewoon prima te functioneren.

[image]

dennissiemensma, 2021-03-20:
Bedankt voor je melding. Gaat het script wel door en krijg je aan het eind iets als `>>> Deployment complete. <<<`?

Zo ja dan voorzie ik verder geen problemen, helemaal als DSMR-reader lijkt te werken. De foutmelding gebeurt namelijk tijdens het installeren van externe libraries.

reboot1983, 2021-03-20:
Sorry, dat had ik er best ff bij kunnen vermelden inderdaad.
Ja, het script is netjes doorgegaan en aan het einde gaf hij aan klaar te zijn en de nieuwe versie is nu 4.13.

dennissiemensma, 2021-03-20:
Oke prima! Ik zal voor de zekerheid een extra check toevoegen aan het deplyscript voor een volgende release. Dank voor je melding!

closed 2021-03-20 by dennissiemensma