dsmrreader/dsmr-reader #1327: remote api toevoegen als uitleesmethode bij Dataloggerconfiguratie

corchat, 2021-03-21:
Hoi,
Wellicht zie ik iets over het hoofd, maar dat hoor ik dan graag van je.
Is het mogelijk om bij de Datalogger configuratie naast Seriele poort en netwerk socket een derde optie remote api te maken.
Hiermee zou dan heel makkelijk bijvoorbeeld de Homewizard P1 kunnen worden uitgelezen. Deze heeft een api waarbij je het volledige telegram kan uitlezen op `http://ipadres/api/v1/telegram`. Het enige verschil met wat dsmr_reader qua telegram is de laatste regel. Het homewizard telegram eindigt op een ! en dsmr_reader op ! gevolgd door 4 tekens.

Als invoervelden denk ik dan aan of 1 complete adresregel of host, poort, pad.
Groet Cornelis

dennissiemensma, 2021-03-21:
Bedankt voor je vraag. Ik begrijp je voorstel, alleen is het helaas momenteel niet mogelijk om elke combinatie te ondersteunen. 

Ik denk dat je dan het beste kunt kijken naar een plugin, zoals in dit voorbeeld met een andere bron, maar soortgelijk verzoek: https://dsmr-reader.readthedocs.io/en/latest/plugins.html#example-5-read-telegrams-using-dsmrloggerws-api

Als ik zo de docs van Homewizard kijk dan staat er een waarschuwing bij die telegram API. Er lijkt echter ook een tweede API te zijn die de gesplitste velden ontsluit, waardoor die checksum (die missende 4 tekens na `!`) niet meer nodig is, het apparaat heeft die dan als het goed is (hoop ik voor je) gecheckt.

```
http://{IP address}/api/v1/data
```

**Kun je eens kijken of je daar data van terugkrijgt? En komt dat overeen met het voorbeeld uit de docs?**

https://energy.homewizard.net/nl/support/solutions/articles/19000117051-homewizard-energy-p1-meter-lokale-api

Want dan kan ik wel even even voor je kijken of ik iets voor je in elkaar kan draaien. Als die plugin werkt dan zal ik hem toevoegen aan de docs, zoals deze: https://github.com/dsmrreader/dsmr-reader/issues/1057#issuecomment-663744705

dennissiemensma, 2021-03-21:
En wat voor slimme meter en DSMR-protocol (v4/5 of ouder) heb je trouwens? Wellicht veroorzaakt dat namelijk die missende checksum.

corchat, 2021-03-21:
Bedankt voor de snelle reactie.
Als ik naar het data pad van de api ga krijg ik dit terug. Dat leek me zekers niet voldoende om dsmr reader mee te vullen.
Helaas ik hoopte op een soort generieke oplossing voor externe api die volledige telegrammen leveren :D)
Ga ik zelf kijken of ik kan knutselen, zonder enige python kennis wordt dat wel een uitdaging 

`
{
  "smr_version": 50,
  "meter_model": "Landis + Gyr $$$$$$$$$$$$",
  "wifi_ssid": "$$$$$$$$$$$$",
  "wifi_strength": 76,
  "total_power_import_t1_kwh": 9999.402,
  "total_power_import_t2_kwh": 6845.332,
  "total_power_export_t1_kwh": 0,
  "total_power_export_t2_kwh": 0,
  "active_power_w": 3068,
  "active_power_l1_w": 3068,
  "active_power_l2_w": null,
  "active_power_l3_w": null,
  "total_gas_m3": null,
  "gas_timestamp": null
}`

dennissiemensma, 2021-03-21:
Bedankt voor je aanvulling. Je hoeft niet perse zelf te knutselen. 

Het enige wat dan nog mist is het tijdstip van de meting, wel vreemd dat ze die niet ontsluiten.

- Wat voor slimme meter en DSMR-protocol (v4/5 of ouder) heb je? 
- Klopt het dat je geen gasmeter aan je slimme meter hebt?
- En kun je ook een telegram uit `http://ipadres/api/v1/telegram` delen? 

Haal voor het laatste wel je serienummer weg, dat is een lang getal op de 3e of 5e regel die begint met:
```
0-0:96.1.1(
```

corchat, 2021-03-21:
Hier is ie, ik heb dsmr5 (ging er vanuit dat dat de smr versie was) en inderdaad alleen electra.
`
/XMX5LGBBLA4415386700

1-3:0.2.8(50)
0-0:1.0.0(210321214450W)
0-0:96.1.1(xxxxxxxxxxxxxxxxxxxxxxxxxx)
1-0:1.8.1(010003.107*kWh)
1-0:1.8.2(006845.332*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(01.992*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00002)
0-0:96.7.9(00000)
1-0:99.97.0(0)(0-0:96.7.19)
1-0:32.32.0(00008)
1-0:32.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(232.0*V)
1-0:31.7.0(008*A)
1-0:21.7.0(01.992*kW)
1-0:22.7.0(00.000*kW)
!
`

corchat, 2021-03-22:
Hoi,
Om het verhaal compleet te maken. Ik heb 'm draaiend.
Eerst de door jou genoemde plugin geïnstalleerd. En toen deed ie nog niks.
Een beetje zitten snuffelen in de broncode en toen las ik dat de checksum die in mijn telegram ontbreekt pas bij dsm versie 4 was ingevoerd. M'n meterversie teruggezet naar dsmr 2/3 en hij loopt als een zonnetje. En aangezien ik alleen electriciteitsverbruik meet acht ik de kans dat ik wat ga missen in voor mij relevante data heel klein.

Ik hoop toch stilletjes dat de remote api ooit een standaard invoermethode voor de datalogger wordt. (dus gewoon vanuit het dataloggerconfiguratie scherm)  Dat zou toch een stuk eenvoudiger zijn. (en voor zover ik kan inschatten is, behalve de invoervelden al aanwezig)
In ieder geval bedankt voor je hulp.
Groet Cornelis

dennissiemensma, 2021-03-22:
Bedankt voor je aanvulling, fijn om te horen! Dat was de reden dat ik vroeg naar je meter, al denk ik dat de fout in Homewizard zit.

Want je telegram geeft aan dat het DSMR v5 is:
```
1-3:0.2.8(50)
```
En dan is het de vraag waarom de checksum mist, want ik kan me niet voorstellen dat je (v5) meter die niet meegeeft. Het kan zijn dat Homewizard de checksum weghaalt na het valideren van het telegram, maar dat is een aanname van mij.
Verder draai je nu een combi van v2/3 parser tegen een v5 meter, dus houd in de gaten of dat goed blijft gaan. Dat het werkt is meer toeval dan bedoeld.

Ik zal het gebruiken van plugins wel stukken simpeler maken via #1328. Dat is het in ieder geval te beheren via de webinterface en maakt het al stukken minder technisch.

closed 2021-03-22 by dennissiemensma

ChristiaanW123, 2022-10-21:
Hoi, is hier ooit nog een follow up voor gekomen? Ik zou graag mijn HomeWizard P1-meter gebruiken met DSMR-reader :).

dennissiemensma, 2022-10-21:
DSMR-reader werkt alleen direct op de meter of per proxy, waarbij een proxy de API van DSMR-reader kan gebruiken om een telegram of de data in DSMR-reader te zetten.
Ofwel het hele telegram via de [v1 API](https://github.com/dsmrreader/dsmr-reader/blob/v5/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py#L87-L92) of de losse (= al geparste) velden via de v2 API.

Andersom is er geen mogelijkheid. Je kunt hooguit `ser2net` gebruiken om je meter uit te lezen. En daar dan zowel DSMR-reader als andere software op laten lezen. Mits andere software ook netwerk-sockets ondersteunt.