dsmrreader/dsmr-reader #1329: Ooit overwogen om prometheus integratie te implementeren?

rogierlommers, 2021-03-21:
Uiteraard biedt dsmr al een mogelijkheid om de "events" naar een influxDB te sturen. Echter sinds prometheus steeds meer tractie begint te krijgen vroeg ik me af: ooit overwogen om een /metrics endpoint te maken? Stelt volgens mij niet zo heel veel voor. In de praktijk komt het op het volgende neer:

- dsmr implementeert een /metrics-endpoint
- dit metrics endpoint wordt "gescraped' door prometheus

That's it! Nu kan iedereen hier met grafana mooie metrics van maken! Ideaal toch?
Zie [hier](https://github.com/jordyv/prometheus-p1-exporter) een voorbeeld van hoe p1 data op een soorgelijke manier exposed wordt.

dennissiemensma, 2021-03-21:
Bedankt voor je vraag. Uiteraard is elke integratie in te bouwen, alleen ben ik erg terughoudend in een wildgroei daarin. Ik zit er niet op te wachten om uiteindelijk na verloop van tijd bijvoorbeeld 15 soorten importers en exporters te moeten onderhouden, en er voor te zorgen dat ze ook allemaal blijven werken bij elke toekomstige wijziging in DSMR-reader. 

Vandaar dat ik het houd bij wat generieke protocollen zoals MQTT (push) en een REST API (pull). Het [eerste beste zoekresultaat](https://github.com/hikhvar/mqtt2prometheus) lijkt al iets te zijn wat MQTT-berichten kan omzetten naar metrics in prometheus. Is zoiets wellicht werkbaar voor je?

rogierlommers, 2021-03-22:
Thanks, @dennissiemensma . Duidelijk verhaal.
Ik zal eens een MQTT lijntje opzetten, dan ben ik er inderdaad ook. 👊

closed 2021-03-22 by dennissiemensma