dsmrreader/dsmr-reader #1331: Upgrade to eCharts 5

dennissiemensma, 2021-03-23:
https://echarts.apache.org/en/tutorial.html#New%20features%20in%20ECharts%205

Los van de update zelf kan dit interessant zijn:
> Time Axis
> Apache ECharts 5 brings a time axis suitable for expressing timestamp scales.

Verder ziet het er iets gelikter uit, subtiele verschillen in rendering, tooltips, etc voor wat ik zo 1-2-3 even zie.

dennissiemensma, 2021-03-28:
Inmiddels goed werkend.

closed 2021-03-28 by dennissiemensma

AbelLykens, 2021-04-05:
Tof. Die negatieve grafieken zijn fijn. Enige onhandige is dat voltagegrafiek y op 0 begint dus dat je schommeling minder ziet :)

dennissiemensma, 2021-04-05:
@AbelLykens bedankt voor je terugkoppeling. 

De voltages worden gefixt in de volgende release. Je kunt het tot die tijd evt handmatig fixen met: https://github.com/dsmrreader/dsmr-reader/issues/1349#issuecomment-812889240