dsmrreader/dsmr-reader #1332: Terugleveren niet duidelijk in grafiek

donnie-dominguez, 2021-03-24:
Sinds update 4.14 zie ik in de live grafieken per fase niet meer duidelijk het verschil tussen het terugleveren en het afnemen van de stroom. Eerder stond er een min-teken voor als ik het me goed herinner.

donnie-dominguez, 2021-03-24:
[image]

dennissiemensma, 2021-03-24:
Bedankt voor je melding. Ik zal het met de volgende release verduidelijken.

----

Kan het overigens zijn dat je dezelfde keluren hebt ingesteld voor teruglevering? Als het goed is wijken die standaard af en anders kun je ze zelf bijwerken op:
```
/admin/dsmr_frontend/frontendsettings/1/change/
```

De standaardkleuren zijn momenteel: `#2E7D32` `#8BC34A` `#9E9D24`

[image]

donnie-dominguez, 2021-03-24:
Ik kan me niet herinneren er ooit iets aan te hebben aangepast. 

De settings staan wel zoals ze default staan, dus wellicht dat daar ook een klein foutje in is geslopen

[image]

dennissiemensma, 2021-03-24:
Hmm dat is vreemd, ik kan het niet reproduceren, afgezien van de +/- die weg is.

Als je op F12 drukt en dan de pagina hard refresht, krijg je daar dan foutmelding of iets te zien onder het tabje "Console"?

----

En als je in de broncode van de pagina kijkt en zoekt op `phase_delivered_l1_color`, krijg je dan wel hetzelfde lijstje met kleurcodes?

[image]

donnie-dominguez, 2021-03-24:
[image]

dennissiemensma, 2021-03-24:
En zie je nog een foutmelding bij tabje `Console` als je F12 doet en dan een refresh?

donnie-dominguez, 2021-03-24:
Mij zegt dit niets :)

[Violation] Added non-passive event listener to a scroll-blocking <some> event. Consider marking event handler as 'passive' to make the page more responsive. See <URL>
echarts.common-4.9.0.min.js:22 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Ht @ echarts.common-4.9.0.min.js:22
gr @ echarts.common-4.9.0.min.js:22
(anonymous) @ echarts.common-4.9.0.min.js:22
D @ echarts.common-4.9.0.min.js:22
fr @ echarts.common-4.9.0.min.js:22
_r @ echarts.common-4.9.0.min.js:22
Cr @ echarts.common-4.9.0.min.js:22
Mr @ echarts.common-4.9.0.min.js:22
wd @ echarts.common-4.9.0.min.js:22
t.init @ echarts.common-4.9.0.min.js:22
(anonymous) @ electricity.js?r=v4.14:2
e @ jquery-3.5.1.min.js:2
t @ jquery-3.5.1.min.js:2
echarts.common-4.9.0.min.js:22 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Ht @ echarts.common-4.9.0.min.js:22
gr @ echarts.common-4.9.0.min.js:22
(anonymous) @ echarts.common-4.9.0.min.js:22
D @ echarts.common-4.9.0.min.js:22
fr @ echarts.common-4.9.0.min.js:22
_r @ echarts.common-4.9.0.min.js:22
Cr @ echarts.common-4.9.0.min.js:22
Mr @ echarts.common-4.9.0.min.js:22
wd @ echarts.common-4.9.0.min.js:22
t.init @ echarts.common-4.9.0.min.js:22
(anonymous) @ phases.js?r=v4.14:3
e @ jquery-3.5.1.min.js:2
t @ jquery-3.5.1.min.js:2
echarts.common-4.9.0.min.js:22 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Ht @ echarts.common-4.9.0.min.js:22
gr @ echarts.common-4.9.0.min.js:22
(anonymous) @ echarts.common-4.9.0.min.js:22
D @ echarts.common-4.9.0.min.js:22
fr @ echarts.common-4.9.0.min.js:22
_r @ echarts.common-4.9.0.min.js:22
Cr @ echarts.common-4.9.0.min.js:22
Mr @ echarts.common-4.9.0.min.js:22
wd @ echarts.common-4.9.0.min.js:22
t.init @ echarts.common-4.9.0.min.js:22
(anonymous) @ power-current.js?r=v4.14:2
e @ jquery-3.5.1.min.js:2
t @ jquery-3.5.1.min.js:2
echarts.common-4.9.0.min.js:22 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Ht @ echarts.common-4.9.0.min.js:22
gr @ echarts.common-4.9.0.min.js:22
(anonymous) @ echarts.common-4.9.0.min.js:22
D @ echarts.common-4.9.0.min.js:22
fr @ echarts.common-4.9.0.min.js:22
_r @ echarts.common-4.9.0.min.js:22
Cr @ echarts.common-4.9.0.min.js:22
Mr @ echarts.common-4.9.0.min.js:22
wd @ echarts.common-4.9.0.min.js:22
t.init @ echarts.common-4.9.0.min.js:22
(anonymous) @ gas.js?r=v4.14:3
e @ jquery-3.5.1.min.js:2
t @ jquery-3.5.1.min.js:2
echarts.common-4.9.0.min.js:22 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
Ht @ echarts.common-4.9.0.min.js:22
gr @ echarts.common-4.9.0.min.js:22
(anonymous) @ echarts.common-4.9.0.min.js:22
D @ echarts.common-4.9.0.min.js:22
fr @ echarts.common-4.9.0.min.js:22
_r @ echarts.common-4.9.0.min.js:22
Cr @ echarts.common-4.9.0.min.js:22
Mr @ echarts.common-4.9.0.min.js:22
wd @ echarts.common-4.9.0.min.js:22
t.init @ echarts.common-4.9.0.min.js:22
(anonymous) @ temperature.js?r=v4.14:3
e @ jquery-3.5.1.min.js:2
t @ jquery-3.5.1.min.js:2
app.min.js?r=v4.14:13 [Violation] 'setTimeout' handler took 138ms

dennissiemensma, 2021-03-24:
Ik heb dan verder geen verklaring waarom het bij jou niet werkt. 

Hoe dan ook ben ik toch al bezig met de upgrade naar de volgende versie van die grafieken library. Draai je DSMR-reader via Docker of heb je het handmatig geinstalleerd? Want ik zoek wat gebruikers om wat varianten uit te proberen.

donnie-dominguez, 2021-03-24:
Het is ook verder geen probleem, wachten we gewoon de nieuwe library af. 
Ik heb hem handmatig geïnstalleerd op een VM met Ubuntu 18.04LTS

dennissiemensma, 2021-03-24:
Heb je interesse om een preview-versie te proberen later deze week? Het gaat in dit geval puur om de visualisatie van de grafieken, de rest werkt hetzelfde.

donnie-dominguez, 2021-03-24:
Zo lang ik niet het risico loop om m'n historische data kwijt te raken (lang leve backups, en snapshots) lijkt het mij prima.  Ik ben dol op overzichtjes en grafieken, dus laat maar komen.

dennissiemensma, 2021-03-25:
Ik heb het inmiddels bijna een dagje draaien en het lijkt goed. Je kunt hier meer informatie vinden en een how-to: https://github.com/dsmrreader/dsmr-reader/issues/1000#issuecomment-805963572

Wil je me sowieso laten weten of die beta-versie direct de **kleuren-issues** fixt van het huidige #1332 ticket? Als dat het fixt dan weet ik dat ik er verder naar hoef te kijken. Dat staat dus los van de experimentele weergave, maar kan het wel als bijeffect oplossen.

dikkeleo2, 2021-03-26:
Hier identiek issue. Sinds de update naar 4.14 zijn de kleuren voor levering en teruglevering in deze grafiek identiek en onafhankelijk van de instelling.

Ik draai vanuit een docker op een pi4.

dennissiemensma, 2021-03-26:
@dikkeleo2 bedankt voor je melding. Ik zal proberen om over een week of iets een nieuwe release te maken, afhankelijk van hoe snel ik #1331 uiteindelijk voor elkaar heb (waar ik dit in mee neem). 
Mochten meerdere grafieken niet goed werken, laat het vooral ook weten.

dikkeleo2, 2021-03-26:
Heel erg bedankt voor uberhaubt alle tijd en moeite die je in dit project steekt!

dennissiemensma, 2021-03-27:
@donnie-dominguez ben je nog in de gelegenheid geweest om de beta-grafieken te proberen?

donnie-dominguez, 2021-03-27:
> @donnie-dominguez ben je nog in de gelegenheid geweest om de beta-grafieken te proberen?

Ik ben net 5 minuten geleden begonnen om een clone te maken van m'n 'productie'-vm om het daar in even te gaan testen.

dennissiemensma, 2021-03-27:
Aah super! Laat tzt maar weten in #1000.

dennissiemensma, 2021-03-27:
Ik sluit het huidige ticket, want in de nieuwe variant straks (wat die ook wordt) komt er iig een duidelijker onderscheid tussen de fasen.

closed 2021-03-27 by dennissiemensma