dsmrreader/dsmr-reader #1334: Foutmelding tijdens installatie van v4.14

tctr74, 2021-03-24:
Ik krijg een error tijdens de upgrade van v4.13 naar v4.14. Zo te zien in het post-deploy script. De foutmelding is bijgevoegd
[deploy.log](https://github.com/dsmrreader/dsmr-reader/files/6201025/deploy.log)

Na het handmatig starten lijkt de webinterface ook niet goed meer te werken. Het huidige verbruik en de live grafieken worden niet weergegeven, maar de logging lijkt verder wel goed te gaan.
[image]
[image]

Debug info dump:
```

DSMR-READER
    App / Python / Database                                                       v4.14 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                    1.0s / 5.0s / 2016h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"
    (!) Database migrations pending                                                                     4 (!)

DATA
    Telegrams total (est.)                                                                            1228300
    Consumption records electricity / gas (est.)                                               135735 / 35497

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 502 MB

```

dennissiemensma, 2021-03-25:
Bedankt voor je melding. **Wat voor OS gebruik je?**

De fout lijkt:
```
ModuleNotFoundError: No module named 'setuptools._distutils'
```

Dit is een tijdje terug ook gemeld in #1307. Je zou die tip kunnen proberen:

```
sudo su - dsmr
export SETUPTOOLS_USE_DISTUTILS=stdlib
./deploy.sh
```

Lost dat het issue voor je op?

tctr74, 2021-03-25:
Hoi,

sudo su - dsmr
export SETUPTOOLS_USE_DISTUTILS=stdlib
./deploy.sh

werkt inderdaad!

Mijn OS is Raspbian 10 op een Pi 3B+.

Thanks!

dennissiemensma, 2021-03-25:
Fijn om te horen, dank voor de terugkoppeling!

closed 2021-03-25 by dennissiemensma