dsmrreader/dsmr-reader #1335: Upgrade faalt op pyyaml

mmartain, 2021-03-25:
(dsmrreader) dsmr@dsmrreader:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
4.14

 --- Checking Python version.
Running Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0]

 --- Checking & synchronizing base requirements for changes.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting autopep8==1.5.6
  Downloading https://www.piwheels.org/simple/autopep8/autopep8-1.5.6-py2.py3-none-any.whl (44 kB)
     |████████████████████████████████| 44 kB 291 kB/s
Requirement already satisfied: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (1.7)
Collecting django==3.1.7
  Downloading https://www.piwheels.org/simple/django/Django-3.1.7-py3-none-any.whl (7.8 MB)
     |████████████████████████████████| 7.8 MB 1.8 kB/s
Collecting django-admin-rangefilter==0.7.0
  Downloading https://www.piwheels.org/simple/django-admin-rangefilter/django_admin_rangefilter-0.7.0-py2.py3-none-any.whl (39 kB)
Requirement already satisfied: django-admin-sortable==2.2.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2.3)
Collecting django-colorfield==0.4.1
  Downloading https://www.piwheels.org/simple/django-colorfield/django_colorfield-0.4.1-py3-none-any.whl (46 kB)
     |████████████████████████████████| 46 kB 453 kB/s
Requirement already satisfied: django-debug-toolbar==3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.2)
Requirement already satisfied: django-filter==2.4.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (2.4.0)
Collecting django-mathfilters==1.0.0
  Downloading https://www.piwheels.org/simple/django-mathfilters/django_mathfilters-1.0.0-py3-none-any.whl (5.9 kB)
Requirement already satisfied: djangorestframework==3.12.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (3.12.2)
Requirement already satisfied: django-solo==1.1.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (1.1.5)
Collecting dropbox==11.4.1
  Downloading https://www.piwheels.org/simple/dropbox/dropbox-11.4.1-py3-none-any.whl (559 kB)
     |████████████████████████████████| 559 kB 1.1 MB/s
Requirement already satisfied: gunicorn==20.0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (20.0.4)
Requirement already satisfied: influxdb==5.3.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (5.3.1)
Requirement already satisfied: paho-mqtt==1.5.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (1.5.1)
Requirement already satisfied: pyserial==3.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.5)
Requirement already satisfied: pyserial-asyncio==0.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (0.5)
Requirement already satisfied: python-dateutil==2.8.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.8.1)
Requirement already satisfied: python-decouple==3.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (3.4)
Collecting pytz==2021.1
  Downloading https://www.piwheels.org/simple/pytz/pytz-2021.1-py2.py3-none-any.whl (510 kB)
     |████████████████████████████████| 510 kB 1.2 MB/s
Collecting pyyaml==5.4.1
  Downloading PyYAML-5.4.1.tar.gz (175 kB)
     |████████████████████████████████| 175 kB 649 kB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
ERROR: Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 171, in _merge_into_criterion
    crit = self.state.criteria[name]
KeyError: 'pyyaml'

[... 49 lines cut ...]
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 86, in _build_backend
    obj = import_module(mod_path)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 953, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 6, in <module>
    import distutils.core
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/_distutils_hack/__init__.py", line 82, in create_module
    return importlib.import_module('._distutils', 'setuptools')
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'setuptools._distutils'

[!] Aborting post-deployment due to PIP
[!] Halted. Post-deployment script exited with non-zero code

dennissiemensma, 2021-03-25:
Wil je #1334 eens proberen?

mmartain, 2021-03-25:
> Wil je #1334 eens proberen?

Jaa! Super dat werkt. Thanks!

closed 2021-03-25 by mmartain