dsmrreader/dsmr-reader #1336: Missing columns after upgrade to v4.14

heinrichnak, 2021-03-26:
Performed an upgrade to v4.14 today and now Live Graphs and Actual Consumption don't work anymore.
I see this in dsmr_backend.log:

```
dsmr_backend.log:[2021-03-26 12:56:32,407] ERROR    (ProgrammingError) <bound method BackupAppConfig._on_backend_called_signal of <BackupAppConfig: dsmr_backup>> errored: column dsmr_backup_dropboxsettings.latest_sync does not exist
dsmr_backend.log:[2021-03-26 12:56:32,417] ERROR    (ProgrammingError) <bound method PvoutputAppConfig._on_backend_called_signal of <PvoutputAppConfig: dsmr_pvoutput>> errored: column dsmr_pvoutput_pvoutputaddstatussettings.next_export does not exist
```

Migration seems to be completed:
```
(dsmrreader) dsmr@ubuntu:~/dsmr-reader$ ./manage.py showmigrations | grep -v '[X]'
admin
auth
contenttypes
dsmr_api
dsmr_backend
dsmr_backup
dsmr_consumption
dsmr_datalogger
dsmr_dropbox
dsmr_frontend
dsmr_influxdb
dsmr_mindergas
dsmr_mqtt
dsmr_notification
dsmr_pvoutput
dsmr_stats
dsmr_weather
sessions
```

Is it just a matter of adding those columns in Postgres?

heinrichnak, 2021-03-26:
Auto fix ;)

Noticed these errors when I re-deployed this version:

``` --- Checking & synchronizing static file changes.
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 194, in handle
    collected = self.collect()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 118, in collect
    handler(path, prefixed_path, storage)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/contrib/staticfiles/management/commands/collectstatic.py", line 355, in copy_file
    self.storage.save(prefixed_path, source_file)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/files/storage.py", line 52, in save
    return self._save(name, content)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages/django/core/files/storage.py", line 267, in _save
    fd = os.open(full_path, self.OS_OPEN_FLAGS, 0o666)
PermissionError: [Errno 13] Permission denied: '/var/www/dsmrreader/static/admin/css/forms.css'```

Fixed the permissions and now it works again.

closed 2021-03-26 by heinrichnak

dennissiemensma, 2021-03-26:
Thanks for reporting and hinting at the cause! I'll make sure to add an additional exit code check to the deploy script next release, to make it more obvious when copying the static files fail. 

And regarding the database errors, they might be caused at the time when the database was updated, but the processes with the old code were still running for a moment. They are softly reloaded at the end of the deploy script, but sometimes it takes a bit or the reload fails.

dennissiemensma, 2021-04-03:
Onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).