dsmrreader/dsmr-reader #1337: Statistics show 'teruglevering' instead of 'verbruik'

reboot1983, 2021-03-26:
When going to the statistics page I see the usage in "HOOGTE- EN LAAGTE RECORDS", but it is stating 'hoogste teruglevering', but we are not returning power to the net. We only take the power. I think there is a wrong label here? It should state 'hoogste verbruik' (per fase). The lowest usage is correct.

Version is the current latest: 4.14
[image]

dennissiemensma, 2021-03-26:
Bedankt voor je melding. Ik heb zelf dezelfde situatie thuis (geen teruglevering) en had er eerder geen verklaring voor. Ik kan me vaag herinneren dat ik het ook echt zo zag staan in de geschiedenis, hoewel dat niet het ruwe telegram was.

[image]

Ik heb het al een paar keer gereset, maar het blijft willekeurig terugkomen, dus ik heb het maar gelaten voor wat het was.

----

Afhankelijk van of en hoever je automatische dataopschoning aan hebt staan, kun je zelf kijken of je nog het betreffende (geparste) telegram van 5 maart kan terugvinden op deze URL:

```
/admin/dsmr_datalogger/dsmrreading/?timestamp__range__gte_0=2021-03-05&timestamp__range__gte_1=14%3A54%3A00&timestamp__range__lte_0=2021-03-05&timestamp__range__lte_1=14%3A55%3A00
```

En dan kijken wat er staat bij:
```
Electricity currently returned:
0.000
Current electricity returned in kW
```

----

Voor echt goed onderzoek bij mezelf zou ik eigenlijk al mijn ruwe telegrammen ergens moeten opslaan en dan hopen dat het komende maanden weer een keertje voor komt, maar vond het zelf wat veel werk om dat te doen.

reboot1983, 2021-03-27:
[image]
[image]

Dit is wat ik heb staan als ik jouw link volg. En Tevens heb ik nog gisteren er even bij gezet. Ik zie overal bij 'returned' 0,000 staan. 
Ik heb ook steekproeven gehouden en alles in de tussenliggende periode (het zijn nogal wat bladzijden, vandaar steekproeven) staat returned op 0,000.

Ik kan me niet herinneren dat ik het eerder heb gezien. Ik heb nog wel een screenshot gevonden dat ik op 13-02-2021 heb gemaakt. Daar zie ik dat de labels wel goed staan (zie hieronder). Mij kennende zat ik toen op de dan recentst uitgebrachte versie.

[image]

reboot1983, 2021-03-27:
Overigens nog een kleine sidenote van deze pagina: de kolomkoppen staan hier in het Engels i.p.v. NL

dennissiemensma, 2021-03-27:
Bedankt voor je aanvulling. De Statistieken-pagina is wel aangepast bij de laatste release, dus ik heb [de template even opnieuw voor je gecheckt](https://github.com/dsmrreader/dsmr-reader/blob/v4.14.0/dsmr_frontend/templates/dsmr_frontend/statistics.html#L227-L271). De mapping lijkt daar wel goed. Als in dat deze velden gebruikt worden:
```
highest_return_l1_value
highest_return_l2_value
highest_return_l3_value
```

Je kunt ze even checken in de database:
```
sudo su - postgres
psql dsmrreader
\x on

select * from dsmr_stats_electricitystatistics;
```

Dan krijg je zoiets:
```
-[ RECORD 1 ]---------------+-----------------------
id                          | 1
highest_usage_l1_timestamp  | 2019-06-09 17:02:20+01
highest_usage_l2_timestamp  | 2019-10-05 11:34:16+01
highest_usage_l3_timestamp  | 2020-12-28 11:57:37+00
highest_return_l1_timestamp | 2020-08-06 20:56:25+01
highest_return_l2_timestamp | 2020-12-02 10:54:02+00
highest_return_l3_timestamp | 2021-02-07 10:48:57+00
highest_usage_l1_value      | 4.301
highest_usage_l2_value      | 6.224
highest_usage_l3_value      | 3.728
highest_return_l1_value     | 0.163
highest_return_l2_value     | 0.095
highest_return_l3_value     | 2.630
lowest_usage_l1_timestamp   | 2019-03-29 16:36:25+00
lowest_usage_l1_value       | 0.006
lowest_usage_l2_timestamp   | 2021-02-19 10:39:13+00
lowest_usage_l2_value       | 0.008
lowest_usage_l3_timestamp   | 2018-11-14 09:30:38+00
lowest_usage_l3_value       | 0.001
```

----

Als de eerdere zoekopdracht via de GUI geen resultaten toont, kun je het snelste via de database checken welke metingen uberhaupt een teruglevering hadden.

Ik zie nu trouwens ook dat de statistieken een ander veld gebruiken, [die van de fasen en niet 'currently returned'](https://github.com/dsmrreader/dsmr-reader/blob/v4.14.0/dsmr_stats/services.py#L367-L377).
```
sudo su - postgres
psql dsmrreader
\x on

select id, timestamp, electricity_currently_returned, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3 from dsmr_datalogger_dsmrreading where phase_currently_returned_l1 > 0 or phase_currently_returned_l2 > 0 or phase_currently_returned_l3 > 0 limit 10;
```

Ik zie nu zelf ook een resultaat op **L3**:
```
-[ RECORD 1 ]------------------+-----------------------
id                             | 13369127
timestamp                      | 2020-04-12 19:00:07+01
electricity_currently_returned | 0.000
phase_currently_returned_l1    | 0.000
phase_currently_returned_l2    | 0.000
phase_currently_returned_l3    | 0.001

```

----

Hoe dan ook is het denk ik goed dat ik het in een volgende release dan maar hardcoded ga verbergen. Op de meeste plekken toont DSMR-reader teruglevering op basis van checks, maar op deze pagina nog niet, zoals je al hebt geprobeerd.

reboot1983, 2021-03-28:
Eerste query:
[image]

Tweede query:
[image]

dennissiemensma, 2021-03-28:
Dank voor de update. Ik zie het al, het is de NL-vertaling. Die klopt niet. In je intiele screenshot staan de kleuren er niet bij, dus valt het me nu pas op. Klopt het dat de kleuren bij teruglevering bij jou **rood** zijn?

dennissiemensma, 2021-03-28:
Aah nvm, in je tweede screenshot staan ze erbij en daar zijn ze idd rood. Ik zal de vertaling fixen, maar hoe dan ook de teruglevering verbergen als die er niet is.

dennissiemensma, 2021-03-28:
Hier kun je het goed zien, ze staan dubbel.

[image]

dennissiemensma, 2021-03-28:
Gefixt en aangepast voor volgende release.

closed 2021-03-28 by dennissiemensma

dennissiemensma, 2021-04-03:
Onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).

reboot1983, 2021-04-03:
Fix confirmed. Zowel teruglevering is nu dynamisch verborgen als de labels zijn nu correct :) (Y)

commit 9508332c (2021-03-28): Translation fix #1337