dsmrreader/dsmr-reader #1338: Https support

Deefnl, 2021-03-27:
Hi,

Is it possible to get https support. Integrated with let's encrypt for example? Right now it's impossible to integrate the web page into a https home assistant iframe.

Regards,

Dave Jacobs

fverpoest, 2021-03-27:
My installation is running without any problem on https with the help of Certbot.

Follow these instructions: https://certbot.eff.org/lets-encrypt/debianbuster-nginx

jeroenpeters1986, 2021-03-27:
Yes, DSMR-Reader works perfectly on HTTPS if you have that configured. This is the part where you will need to configure your webserver to server DMSR-reader over HTTPS with a valid certificate. This kinda falls outside the DSMR-Reader scope I think, but running it over HTTPS is working fine

dennissiemensma, 2021-03-27:
Thanks guys, I'll add a reference to the website above in the docs

dennissiemensma, 2021-03-28:
Added to docs for next release.

closed 2021-03-28 by dennissiemensma

dennissiemensma, 2021-04-03:
Onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).