dsmrreader/dsmr-reader #1339: Datepicker is out-of-date

reboot1983, 2021-03-27:
[image]

Kijkende naar de datepicker. Als je hier iets wilt selecteren ben je je een ongeluk aan het klikken naar het huidige jaar. :)
Kan de ingevulde datum hier worden preselected?

dennissiemensma, 2021-03-27:
Bedankt voor je melding. Kun je eens kijken of het ook gebeurt als je gewoon `/admin/dsmr_datalogger/dsmrreading/` opent?

Ik kan het namelijk alleen reproduceren als ik de pagina benader via een URL met een voorgedefinieerde zoekopdracht. Daar lijkt die minder goed mee om te gaan en dat is mij niet eerder opgevallen. Hoe dan ook [is dit een externe plugin](https://github.com/silentsokolov/django-admin-rangefilter), dus ik kan maar beperkt wijzigingen hier omheen doen.

reboot1983, 2021-03-28:
Dan ziet hij er inderdaad wel goed uit

dennissiemensma, 2021-03-28:
Dank voor het proberen! In dat geval laat ik het voor nu voor wat het is, gezien ik vermoed dat vrij weinig gebruikers deze specifeke situatie meemaken.

closed 2021-03-28 by dennissiemensma