dsmrreader/dsmr-reader #1340: Too many unprocessed readings: 971012   (2 months, 1 week ago)

Paul-dH, 2021-03-28:
Goede morgen!

Ik heb DSMR een poosje laten lopen (tussentijds is het wat druk geweest) en blijkbaar heeft hij veel opgehoopt aan readings. Hij geeft dit bericht aan en ik heb geprobeerd om in de container de backend te herstarten. Ook is de container herstart en inmiddels van 4.9 naar 4.14 gemigreerd. Echter blijft het bericht staan...

Op het Dashboard wordt wel live aangegeven wat het verbruik is, maar op de live chart is niets te zien.

Wat kan ik doen om dit te troubleshooten?

Thanks, ene fijne zondag! :innocent:

dennissiemensma, 2021-03-28:
Bedankt voor je vraag. Je kunt het beste beginnen bij de logiles en dan specifiek die van het backend-proces: https://dsmr-reader.readthedocs.io/nl/latest/how-to/troubleshooting/logfiles.html

Als je daar niets in kan vinden, probeer DEBUG-logging tijdelijk aan te zetten om te zien wat DSMR-reader doet en waarom die de metingen niet verwerkt: https://dsmr-reader.readthedocs.io/nl/latest/how-to/troubleshooting/enabling-debug-logging.html

dennissiemensma, 2021-03-28:
Na het inschakelen van DEBUG-logging kun je dus weer die logfile van de backend opnieuw bekijken, ik zal dat nog wat duidelijker vermelden in de docs.

Paul-dH, 2021-03-30:
Hoi Dennis,

Na een dagje klussen in de tuin heeft het probleem zich toch vanzelf opgelost. De herstart van de container met de upgrade moet hiervoor de trigger geweest zijn. Het was voor 2,5 maanden aan datapunten wat hierna netjes verwerkt werd. Ik ben blij dat alles er toch nog in staat :)

Bedankt voor de mooie software en sorry voor het loze issue.

Groetjes, Paul

closed 2021-03-30 by Paul-dH

dennissiemensma, 2021-03-30:
Fijn om te horen en prima hoor!