dsmrreader/dsmr-reader #1341: Hoe kan ik backup van database terug zetten in Homeassistant

mgiebels, 2021-03-28:
Hoe kan ik backup van database terug zetten in Homeassistant?

Ik ben van een Pi naar NUC gegaan en nu ben ik me data kwijt kan ik dit herstellen?

dennissiemensma, 2021-03-28:
Bedankt voor je vraag. Het is mij niet helemaal duidelijk wat je bedoelt. Wil je gegevens die DSMR-reader ooit naar HA heeft gestuurd terugzetten in HA?

mgiebels, 2021-03-28:
Hoi Dennis ja ik wil me data die ik op me PI had terug zetten op me NUC ik had in de map backup/dsmrreader alles staan (tar bestanden) alleen zoek ik hoe ik dat terug kan zetten

dennissiemensma, 2021-03-28:
Heb je een lijstje van de bestanden die je bedoelt? Er is een kans dat het namelijk over iets anders gaat, want ik weet niet meer zeker of ik ooit `tar` bij DSMR-reader heb gebruikt.

Als het een databasebackup van DSMR-reader zelf is dan kun je hier vinden hoe: https://dsmr-reader.readthedocs.io/nl/latest/how-to/database/postgresql-restore-backup.html
Voor de duidelijkheid: dat verandert niets in HA, alleen DSMR-reader zelf.

mgiebels, 2021-03-28:
[image]
Deze bestanden zijn het GZ bestanden geen tar excuus maar die handleiding werkt niet onder HA die commando's kan je niet uitvoeren namelijk

dennissiemensma, 2021-03-28:
Ah precies, dit zijn idd DSMR-reader backups. DSMR-reader en HA zijn twee hele losse applicaties en niet uitwisselbaar met elkaar, dus je zult de backups moeten importeren in DSMR-reader. 
Wellicht heb (of had) je DSMR-reader op dezelfde Pi geinstalleerd? Heb je die al weer opnieuw geinstalleerd?

mgiebels, 2021-03-28:
Yes heb ik alweer geinstalleerd ik dacht simpel zet ze in de backup maps maar dat werkt niet ;)
Daarom ook de hulp vraag hoe importeer ik die in DSMR reader

dennissiemensma, 2021-03-28:
Je kunt de DB-backup alleen importeren in een lege database. Daarvoor zul je de nieuwe (bijna lege) database van DSMR-reader moeten verwijderen. 
Klopt het dat je vanochtend om iets voor 11:00 nog een handmatige backup hebt gemaakt? Wellicht voordat je de herinstallatie deed? Gezien die tijden van de sunday backup iets afwijken van de rest?

mgiebels, 2021-03-28:
Ja klopt wel ja dat die toen gedraaid heeft

dennissiemensma, 2021-03-28:
Oke dan kun je dit proberen als sudo (meestal `pi`) of `root` user:

Dit **verwijdert** dus wel je nieuwe database, met vermoedelijk slechts een uur aan gegevens:
```
sudo supervisorctl stop all

sudo -u postgres dropdb dsmrreader
sudo -u postgres createdb -O dsmrreader dsmrreader

sudo zcat /home/dsmr/dsmr-reader/backups/dsmrreader-postgresql-backup-Sunday.sql.gz | sudo -u postgres psql dsmrreader
```

Let goed op of je geen fouten krijgt ergens.

Dan kun je DSMR-reader weer aanzetten:

```
sudo su - dsmr
./deploy.sh

logout
sudo supervisorctl start all
```

mgiebels, 2021-03-28:
Sudo kent hij niet als ik ssl naar me Homeassistant dan geeft hij sudo command not found

dennissiemensma, 2021-03-28:
Je zult dit op de Pi zelf moeten doen, niet in HA. Hoe heb je DSMR-reader geinstalleerd?

mgiebels, 2021-03-28:
DSMR reader heb ik als add on geinstalleerd vanuit Homeassistant

dennissiemensma, 2021-03-28:
Toevallig [deze add-on](https://community.home-assistant.io/t/dsmr-reader-add-on-for-home-assistant/279087)? Zo ja, dan kun je het het beste bij de ontwikkelaar van die add-on leggen: [issue aanmaken](https://github.com/sanderdw/hassio-addons/issues). 
En daarbij kun je eventueel linken naar het huidige issue, voor alle commands hierboven die normaal uitgevoerd moeten worden.

Het probleem is namelijk dat ik niet weet hoe die add-ons werken en hoe je bij de database komt daar. Waarschijnlijk is het onderwater Docker, maar ook daar kan ik je zeer beperkt bij helpen.

dennissiemensma, 2021-03-28:
Voor nu sluit ik het issue. Echter, voel jullie uiteraard vrij om later hier alsnog verdere vragen of info te achterhalen, indien nodig!

closed 2021-03-28 by dennissiemensma

mgiebels, 2021-03-28:
geen probleem dennis dank je wel even goed