dsmrreader/dsmr-reader #1342: Timestamp from DSMR5 meter is misinterpreted

svloosbroek, 2021-03-29:
/Ene5\T210-D ESMR5.0

1-3:0.2.8(50)
0-0:1.0.0(210329232030S)
...
ID:
1
Timestamp:
30 maart 2021 01:20
Moment waarop de meting is gedaan

The above is a copy-paste from the meter statistics admin page. There's a 2 hour discrepancy between the timestamp from the telegram and how it is interpreted. Runs in Docker with the - DJANGO_TIME_ZONE=Europe/Amsterdam flag. What can clarify this?

dennissiemensma, 2021-03-29:
Bedankt voor je melding. Grote kans dat je van het volgende last hebt: #1282

Probeer deze mount te verwijderen:
```
/etc/localtime:/etc/localtime:ro
```

svloosbroek, 2021-03-30:
Dat was inderdaad de fix, dnak. Geen idee hoe die er is gekomen, in de laatste versie van het docker compose voorbeeld staat-ie er ook niet.

closed 2021-03-30 by svloosbroek