dsmrreader/dsmr-reader #1343: Upgrade naar v4.14; geen Consuming-data meer

AbelLykens, 2021-03-30:
Vanochtend geupgrade naar v4.14, menuutje zegt:

```
 Currently installed: DSMR-reader v4.14
 Everything seems to run smoothly. Any issues, such as missing data or stale processes, will be reported here.
```

Er lijkt wat mis te gaan met de processen:
``` supervisord[492]: 2021-03-30 06:59:44,503 INFO exited: dsmr_backend (exit status 0; expected)
 supervisord[492]: 2021-03-30 06:59:45,214 INFO spawned: 'dsmr_backend' with pid 874
 supervisord[492]: 2021-03-30 06:59:45,215 INFO exited: dsmr_datalogger (exit status 0; expected)
 supervisord[492]: 2021-03-30 06:59:45,718 INFO spawned: 'dsmr_datalogger' with pid 877
 supervisord[492]: 2021-03-30 06:59:47,213 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
 supervisord[492]: 2021-03-30 06:59:47,214 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
 supervisord[492]: 2021-03-30 07:00:18,371 INFO exited: dsmr_backend (exit status 0; expected)
 supervisord[492]: 2021-03-30 07:00:19,373 INFO spawned: 'dsmr_backend' with pid 892
 supervisord[492]: 2021-03-30 07:00:20,873 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
 supervisord[492]: 2021-03-30 07:00:52,636 INFO exited: dsmr_backend (exit status 0; expected)
 supervisord[492]: 2021-03-30 07:00:53,638 INFO spawned: 'dsmr_backend' with pid 903
 supervisord[492]: 2021-03-30 07:00:55,195 INFO success: dsmr_backend entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

Ik draai op `Debian GNU/Linux 10 (buster)` in een python3 virtualenv.

dennissiemensma, 2021-03-30:
Bedankt voor je melding. De issues met tijdzones zijn alleen van toepassing als je Docker gebruikt. 

- Het is mij niet helemaal duidelijk op welke DSMR-reader versie je eerder zat. Was dat al een 4.x of nog een oudere versie?
- Heb je ook het [standaardscript](https://dsmr-reader.readthedocs.io/nl/latest/how-to/upgrading/upgrade.html) voor upgraden uitgevoerd? Zo nee, dan kan ik je aanraden om altijd die uit te voeren ipv losse commands. 
- En als je eerdergenoemd script nu nogmaals uitvoert, zie je dan ergens foutmeldingen?

dennissiemensma, 2021-03-30:
Verder nog de tip om altijd te kijken in de logfiles van de individuele processen: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/logfiles.html

Ik verwacht dat je daar een duidelijke(re) foutmelding terugziet.

AbelLykens, 2021-03-30:
Thanks voor je reply

Vandaag leerde ik over deploy. Ik had die stappen zelf bij elkaar gezocht, zag bij de releasepagina geen instructies... 

Misschien een linkje vanaf de hoofdrepo? Ik klikte op installatieinstructies en kreeg niet wat ik zocht.

[image]

Dank voor meedenken, hij doet het weer na het draaien van deploy!

closed 2021-03-30 by AbelLykens