dsmrreader/dsmr-reader #1344: Weergrafieken op archive pagina

thijskuijpers, 2021-04-01:
De archive pagina bevat momenteel grafieken voor elektriciteit en gas. Is het mogelijk om hieronder ook een line/bar chart op te nemen met de gemiddelde temperatuur over de periodes?

dennissiemensma, 2021-04-01:
Bedankt voor je verzoek! Hoewel ik je wens goed begrijp, heb ik eerder besloten om geen verdere features meer te ondersteunen qua weergegevens. 

Achteraf gezien was het namelijk een slechte keus om die op te nemen in DSMR-reader omdat het niets met DSMR te maken heeft en omdat de temperatuur alleen, zoals die gemeten is, niet afdoende is om er daadwerkelijk iets aan af te leiden. DSMR-reader focust meer op het verbruik zelf, ook om het wiel niet opnieuw te hoeven uitvinden.

Je kunt het beste kijken naar MinderGas.nl, die ook daadwerkelijk rekening houdt met graaddagen voor vergelijkingen: https://www.mindergas.nl/degree_days/explanation

- #1200
- #426
- #738
- #1279

closed 2021-04-01 by dennissiemensma

thijskuijpers, 2021-04-02:
Dank voor je suggestie. Mindergas gaat echter alleen over gas, ik ben juist geinteresseerd in het effect van temperatuur op het stroomverbruik. Ik zal op zoek gaan naar een alternatief hiervoor.