dsmrreader/dsmr-reader #1346: Dark mode

AbelLykens, 2021-04-01:
* Dark screens use less power!
* Dark screens read easier!
* Dark mode is appreciated by many many people

Would be nice to have a dark mode!

dennissiemensma, 2021-04-01:
Bedankt voor je verzoek. Ik zal hier ooit een keer naar kijken, maar ik verwacht daar niet op korte termijn wat voor. 

Het is namelijk niet een kwestie van even kleuren wisselen, het vereist soms echt een compleet andere aanpak qua vormgeving. En ik ben zelf geen ook designer, dus ik heb er niet zomaar een opzet voor.

*Ik denk overigens niet dat elk scherm perse minder verbruikt bij zwarte kleuren, dat hangt echt van het type scherm af, maar dat is een andere discussie ;-)*

jimmyd-be, 2021-05-27:
Op de desktop kan ik "Dark reader" addon voor Chrome browsers (chromium, Chrome, Edge) aanraden. Deze maakt DSMR reader in een mooi dark theme.
Ik wil gerust dit bekijken om in de app zelf te steken. Maar kan niets beloven van tijdschema wanneer ik er naar kan kijken.

dennissiemensma, 2021-05-27:
Bedankt voor je tip! Als daar mooie kleuren uit komen dan zou ik dat evt kunnen gebruiken als basis.

Ik zie nu ook dat er een `@media (prefers-color-scheme: dark) {` CSS hint is, die dan ook van pas kan komen, maar dan voor de toggle zelf: https://css-tricks.com/a-complete-guide-to-dark-mode-on-the-web/

dennissiemensma, 2022-02-04:
Dit wordt vermoedelijk opgepakt in v5.1, via een PR van [Justin991q](https://github.com/Justin991q): #1567

dennissiemensma, 2022-02-19:
Komt in v5.1

closed 2022-02-19 by dennissiemensma