dsmrreader/dsmr-reader #1347: importeren partial backup

casrenooij, 2021-04-01:
Hi,

Ik heb eerder een herinstallatie gedaan van DSMR en dat lukte allemaal.  Nu probeer ik het weer maar nu kom ik in de problemen  na :
 zcat <PATH-TO-POSTGRESQL-BACKUP.sql.gz> | sudo -u postgres psql dsmrreader  ( met het correcte pad :)  )

Zie de output onderaan :

Om er zeker van te zijn dat ik alle stappen exact heb uitgevoerd zoals ze op de "https://dsmr-reader.readthedocs.io/nl/v4/how-to/database/postgresql-restore-backup.html"  pagina staan heb ik de hele server ( op ESX ) drie keer totaal van scratch opnieuw opgebouwd.  Mijn eerdere poging ging vlekkeloos dus ik weet dat het kan. Het enige verschil is dat er tussentijds een update is geweest.  

Wat kan de oorzaak zijn, en belangrijker, wat is de oplossing ?

```
 psql dsmrreader
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
SET
ERROR:  relation "public.dsmr_stats_daystatistics" does not exist
invalid command \N
<KNIP heleboel regels >
invalid command \N
invalid command \.
ERROR:  syntax error at or near "1"
LINE 1: 1 2021-01-03 0.00 14.193 0.000 0.068 0.000 0.00 0.00 6.504 0...
        ^
invalid command \.
ERROR:  syntax error at or near "1"
LINE 1: 1 2021-01-03 10:00:00+01 0.478 0.000 0.000 0.000 0.034
        ^
ERROR:  relation "public.dsmr_stats_hourstatistics_id_seq" does not exist
LINE 1: SELECT pg_catalog.setval('public.dsmr_stats_hourstatistics_i...
                                 ^
```

Verder heel blij met DSMR, minder met de hardware waar ik het tot nu toe op geinstalleerd heb.

Vrolijk pasen alvast !

dennissiemensma, 2021-04-01:
Bedankt voor je vraag. De partial backups zijn relatief lastig, omdat ze niet in een lege database **zonder structuur en data** kunnen, maar wel in een database die **alleen de structuur** heeft zonder data.

Als je database inderdaad nog leeg is, in de zin dat DSMR-reader niet draait en vooral geen dag/uur-data erin gezet heeft, kun je dit proberen:
```
sudo su - dsmr
./migratie
logout
```

Probeer nu opnieuw die import.

dennissiemensma, 2021-04-01:
De toekomstige pagina van backup terugzetten heeft overigens wat meer informatie, maar de partials missen daar nog in dus die zal ik later nog toevoegen hier:
https://dsmr-reader.readthedocs.io/nl/development/how-to/database/postgresql-restore-backup.html

casrenooij, 2021-04-02:
Ik ben iets verder ( denk ik )

Ik heb de from scratch installatie helemaal afgemaakt zonder import.
daarna :
''''
cas@dsmr:~$ sudo supervisorctl stop all
cas@dsmr:~$ sudo -u postgres dropdb dsmrreader
cas@dsmr:~$ sudo -u postgres createdb -O dsmrreader dsmrreader
cas@dsmr:~$ sudo su - dsmr
(dsmrreader) dsmr@dsmr:~/dsmr-reader$ ./deploy.sh

'''
daarna dit : 
'''
cas@dsmr:~$ zcat  /home/cas/dsmrreader-postgresql-partial-backup-2021-03-29.sql.gz | sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
SET
COPY 82
COPY 1944
 setval
--------
     82
(1 row)

 setval
--------
   1944
(1 row)

cas@dsmr:~$ sudo supervisorctl start all

dsmr_datalogger: started
dsmr_backend: started
dsmr_webinterface: started
cas@dsmr:~$
'''
import leek te werken maar archief is leeg.

Wat kan ik nog proberen ?

''''

dennissiemensma, 2021-04-02:
Je zou kunnen kijken of het importeren in de database ook echt gelukt is, of je daar wel gegevens ziet:

```
sudo su - postgres
psql dsmrreader

select id, day, electricity1, electricity2 from dsmr_stats_daystatistics order by day desc;
```

dennissiemensma, 2021-04-02:
De docs zijn overigens opnieuw bijgewerkt, nu wat vollediger voor een volgende keer: https://dsmr-reader.readthedocs.io/nl/development/how-to/database/postgresql-restore-backup.html

dennissiemensma, 2021-04-03:
Nu ook onderdeel van de nieuwe [v4.15 release](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.0).

dennissiemensma, 2021-04-04:
@casrenooij is dit inmiddels gelukt?

casrenooij, 2021-04-05:
> @casrenooij is dit inmiddels gelukt?

Hoi Dennis, ik heb de installatie opnieuw gedaan, de lege database bekeken, email log geimporteerd en nog steeds niets in het archief. Toen viel mij een regeltje op namelijk dat er nog geen readings binnen waren gekomen. Dat klopte want ik had mijn noodlaptopje waar ik tijdelijk dsmr op heb staan aan de p1 kabel hangen. 
Dus de kabel omgeprikt en bekeken of er data binnen  kwam.  ( was niet zo, dat lag een brltty die tegenwoordig metDebian geinstalleerd wodt ). Toen er eenmaal data binnen kwam, de backup ingelezen en voila.  het werkte.  Niet geheel volgens jouw instructies maar het werkt. Bedankt weer voor je ondersteuning.

Cas..

dennissiemensma, 2021-04-05:
Oke fijn om horen!

closed 2021-04-05 by dennissiemensma