dsmrreader/dsmr-reader #1348: Mogelijkheid voor vergelijken van weken en kwartalen

reboot1983, 2021-04-03:
In het menu 'Vergelijk' kunnen we nu dagen, maanden en jaren vergelijken. Is het ook mogelijk om hier weken en kwartalen aan toe te voegen? Dit zou analytisch een heel mooie uitbreiding zijn.

dennissiemensma, 2021-04-03:
Bedankt voor je suggestie. Ik kan me er goed in vinden, alleen weet ik niet of het mogelijk is met de datepicker die gebruikt wordt. 

De zooming levels daar hebben [geen mogelijkheid voor weken of kwartalen](https://bootstrap-datepicker.readthedocs.io/en/stable/options.html#maxviewmode) en de datepicker lijkt ook niet meer actief onderhouden te worden. 
Uiteindelijk is er misschien wel een workaround voor te maken, alleen verwacht ik dat dat dan een paar keer zoveel werk is en ik weet niet of dat het waard is. Dus voor nu houd ik het even op de backlog, maar weet ik niet zeker of ik er (ooit) ruimte voor heb.

bockelaar, 2021-05-19:
idem om meerdere jaren bijvoorbeeld op week of maand niveau te vergelijken. Denk aan alle maanden mei van de afgelopen 7 jaar naast elkaar. Een flexibele, uitbreidbare vergelijkingsmodule dus :)

dennissiemensma, 2021-09-20:
Ik heb hier nogmaals naar gekeken en ik denk dat hiervoor andere tools beter te gebruiken zijn. Andere tools waarbij ik vooral doel op de opslag en generieke visualisatie van (een hoop) gegevens via een time-series database, ipv een relationele database zoals in DSMR-reader.

Ik denk dat ik de huidige zooming levels in DSMR-reader houd zoals ze zijn, omdat het relatief veel werk is om daar andere levels aan te voegen die niet mappen met de gebruikte 3rd party date picker, zonder daar weer iets anders voor te moeten vinden.
Uiteraard is alles (uiteindelijk) te maken, maar voor mijn gevoel zou ik dan iets maken waar iets als InfluxDB/ElasticSearch i.c.m. iets als Grafana veel beter in zijn. 

Zoals ik het zie biedt DSMR-reader een eenvoudige opslag en weergave van metergegevens. Ik besteed dan liever tijd aan bijvoorbeeld een integratie van export naar InfluxDB 2.x of kwartiergemiddelden voor verbruik/teruglevering, waar Belgie volgend jaar al mee te maken krijgt en NL wellicht over een paar jaar (wie weet).

closed 2021-09-20 by dennissiemensma

dennissiemensma, 2021-09-22:
Nog ter aanvulling (ook wegens referenties vanuit andere issues), het is belangrijk om te weten dat ik de afgelopen jaren steeds meer kijk naar wat er uberhaupt beschikbaar is aan tools.

DSMR-reader is nu bijna 6 jaar geleden opgezet en in 6 jaar is een hoop veranderd qua tools en ook wat gebruikers willen/verwachten. Het uitlezen van meters is stukken makkelijker geworden (kant en klare kits die op wifi werken) en er zijn ook veel meer aanbieders (zowel als online dienst als tools die lokaal te installeren zijn), dan al niet gecombineerd of afhankelijk van eerder genoemde kits. Dat maakt voor DSMR-reader en wat het nu doet inhoudelijk niet uit, maar wel voor de scope van eventuele features. 

Want wat een belangrijkste verschil is dat als ik DSMR-reader vandaag opnieuw zou bouwen, ik waarschijnlijk uitkom op een timeseries database (a la InfluxDB/ElasticSearch) met daar overheen wat dashboading (Grafana?) plus mogelijk nog een schil met features. 
Zonder uberhaupt te definieren wat er gevisualiseerd wordt, kunnen gebruikers daar al ontzettend veel mee. En als er iets niet standaard geleverd wordt, zijn de gegevens in timeseries ongetwijfeld op zo'n manier te aggregeren dat het wel mogelijk is. Zonder afhnakelijk te zijn van een ontwikkelaar.

Ik denk dat dat ook meer van deze tijd is, minder vendor lock, meer community gedreven. Dus dan help ik liever mee om (voor de gebruikers die daar behoefte aan hebben) features te maken die het spiegelen van een timeseries database versimpelt. Dat zeggende hebben zie ik in DSMR-reader geen rol in het ondersteunen van tig soorten exports, want daarvoor is het beter om simpelweg een scriptje te maken die bijvoorbeeld alleen [de parser](https://github.com/ndokter/dsmr_parser) gebruikt + whatever timeseries client nodig is. Daar is namelijk een compleet project op zichzelf aan te wijden.