dsmrreader/dsmr-reader #1349: 4.15 voltage grafiek autoscaling in live grafieken

reboot1983, 2021-04-03:
Sinds versie 4.15 zijn de grafieken in de live-grafieken-pagina herzien. Er zijn mooie opties bijgekomen die in de meeste grafieken heel mooi uitwerken. Echter het voltage gedeelte is nu niet langer inzichtelijk (eigenlijk: nauwkeurig genoeg weergegeven). Dit komt omdat de grafiek niet langer auto-scaled, maar nu tussen de 0 en 250v wordt weergegeven. Hierdoor is het wisselende voltage niet langer duidelijk zichtbaar. Op mijn aansluiting wisselt het voltage soms met +/- 10v en met de huidige schaal is dit niet langer zichtbaar of op te merken.

Versie <4.15
[image]

Versie 4.15
[image]

dennissiemensma, 2021-04-03:
Bedankt voor je melding! Dat is helaas een bijeffect geweest van de grote refactoring, dus ik zal het meenemen in de volgende release.

Tot die tijd kun je evt handmatig de fix doorvoeren: 9ff735cc6ca15b3b5ce9790e362c206d8a1481a6

Let op dat je de fix dan doorvoert op deze file hieronder en **niet** die uit de repo in `/home/dsmr/dsmr-reader/...`:
```
/var/www/dsmrreader/static/dsmr_frontend/js/dsmr-reader/live/voltage.js
```

Na een harde refresh in je browser zou het dan gefixt moeten zijn, totdat je ooit weer `deploy.sh` draait (die overschrijft de file als het goed is).

closed 2021-04-03 by dennissiemensma

dennissiemensma, 2021-04-05:
Fix uitgebracht in [v4.15.1](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.1)