dsmrreader/dsmr-reader #1350: Energiekosten rariteit.

cmderks93, 2021-04-03:
[image]
Ik ben benieuwd hoe deze rekensom tot stand komt, er lijkt iets niet te kloppen.

cmderks93, 2021-04-03:
[image]

dennissiemensma, 2021-04-03:
Bedankt voor je melding. Het is mij niet duidelijk wat je bedoelt. Kun je aangeven wat er precies mis gaat en wat de verwachte uitkomst is? 

Verder kun je op de pagina met energiecontract-totalen beter zien welke prijzen er ingesteld zijn.

cmderks93, 2021-04-03:
Nu ik het uitschrijf, zie ik dat ik een fout maak...
Ik kan ofwel niet tellen, ofwel niet goed kijken....
Deze mag dicht. Sorry!

dennissiemensma, 2021-04-03:
Bedankt voor de opheldering. Meld het vooral als je alsnog ergens een foute berekning ziet hoor.

closed 2021-04-03 by dennissiemensma

dennissiemensma, 2021-04-03:
Ik zal voor de volgende release de (sub)totalen lichtjes highlighten, zodat wat duidelijker is dat het totalen zijn. Ik vermoed dat o.a. daar je verwarring door ontstond.

[image]