dsmrreader/dsmr-reader #1351: Foutmelding bij het opnieuw berekenen van prijzen met Home Assistant addon

TychoWerner, 2021-04-04:
Ik kwam er achter dat ik mijn kWh tarief verkeerd had ingevuld en wilde deze veranderen. Nu las ik dat je bepaalde commando's moet doen om opnieuw te laten bereken. Volgens pagina: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/recalculate-prices.html

Nu draai ik DSMR-reader in Home Assistant als add-on dus het was wat lastig om bij de docker container te komen. Nu heb ik uiteindelijk via Portainer bij de console gekomen. Nu zie ik wel dat ik de goede map zit door ```ls``` te doen en dan zie ik de bestanden met de belangerijke ```./manage.py```

Nu wilt de het eerste commando ```sudo su - dsmr``` dat ik wissel naar de dsmr gebruiken maar volgens het commando ```/etc/passwd``` bestaat deze niet. Hierdoor faalt het commando dus ook. Nu weet ik niet of dit ook het probleem is voor de volgende commando.

```./manage.py dsmr_stats_recalculate_prices``` nu probeerde ik dit en kreeg het volgende in mijn scherm als melding
```
/dsmr # ./manage.py dsmr_stats_recalculate_prices
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/local/lib/python3.9/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
psycopg2.OperationalError: could not translate host name "dsmrdb" to address: Name does not resolve

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/dsmr/./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/usr/local/lib/python3.9/site-packages/django/core/management/__init__.py", line 395, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/base.py", line 330, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/usr/local/lib/python3.9/site-packages/django/core/management/base.py", line 371, in execute
    output = self.handle(*args, **options)
  File "/dsmr/dsmr_stats/management/commands/dsmr_stats_recalculate_prices.py", line 11, in handle
    dsmr_stats.services.recalculate_prices()
  File "/dsmr/dsmr_stats/services.py", line 408, in recalculate_prices
    for current_day in DayStatistics.objects.all():
  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 287, in __iter__
    self._fetch_all()
  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 1308, in _fetch_all
    self._result_cache = list(self._iterable_class(self))
  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 53, in __iter__
    results = compiler.execute_sql(chunked_fetch=self.chunked_fetch, chunk_size=self.chunk_size)
  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/compiler.py", line 1154, in execute_sql
    cursor = self.connection.cursor()
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 259, in cursor
    return self._cursor()
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 235, in _cursor
    self.ensure_connection()
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 219, in ensure_connection
    self.connect()
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/base/base.py", line 200, in connect
    self.connection = self.get_new_connection(conn_params)
  File "/usr/local/lib/python3.9/site-packages/django/utils/asyncio.py", line 26, in inner
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/django/db/backends/postgresql/base.py", line 187, in get_new_connection
    connection = Database.connect(**conn_params)
  File "/usr/local/lib/python3.9/site-packages/psycopg2/__init__.py", line 127, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)
django.db.utils.OperationalError: could not translate host name "dsmrdb" to address: Name does not resolve
```

Het gaat trouwens om Add-on versie 0.4.0
Hopelijk kan iemand mij hier mee helpen, als er info mist hoor ik het graag

Alvast bedankt!

dennissiemensma, 2021-04-04:
Bedankt voor je vraag. Het lijkt erop dat DSMR-reader (op de commandline) de database niet kan vinden, via de interne Docker host `dsmrdb`.

Ik weet zelf ook niet precies hoe je dit kan oplossen bij de setup die je gebruikt. Ik denk dat je dit het beste eerst bij de ontwikkelaar van de betreffende HA-addon zelf kan vragen.

dennissiemensma, 2021-04-04:
Mocht je uiteindelijk verder komen maar inhoudelijk tegen een andere foutmelding aanlopen, voel je vooral vrij om hier later nog follow-up te vragen.

closed 2021-04-04 by dennissiemensma