dsmrreader/dsmr-reader #1352: Geen archief gasmetingen zichtbaar, terwijl Elektra wel zichtbaar is

tdn131, 2021-04-05:
Vanaf 01-01-2017 ben ik elektra en gas in Domoticz gaan registreren. Per 10-7-2020 ben ik overgegaan naar DSMR. Mijn archief gas verbruik is echter niet zichtbaar in DSMR, terwijl het wel op enkele plekken in de database staat. Met het archief elektra is het importeren wel goed gegaan. Gas en elektraverbruik na 10-07-2020 (via DSMR) werkt probleemloos.

Ik heb het volgende script gebruikt om de oude data in DSMR te importeren: https://github.com/dsmrreader/dsmr-reader/issues/908#issuecomment-592456012

Ik heb de volgende tabellen geleegd:
dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption
dsmr_stats_hourstatistics
dsmr_stats_daystatistics
dsmr_datalogger_dsmrreading (alleen readings van 1-1-2017 t/m 09-07-2020)

Vervolgens heb ik opnieuw geïmporteerd. Helaas krijg ik het gasverbruik niet zichtbaar. Ik heb het idee dat er wat onderwater misgaat. Zie de bijgevoegde screenshots, waar ik bijzonderheden op noteer.
1 Import data in dsmr_datalogger_dsmrreading
[image]

2 Resultaat dsmr_consumption_gasconsumption
[image]

3 Resultaat in dsmr_consumption_electricityconsumption (werkt perfect in archief, geen problemen)
[image]

4 Statistieken in dsmr_stats_daystatistics
[image]

Ik gebruik xirixiz/dsmr-reader-docker:latest op Home Assistant OS 64bit op een Raspberry Pi4 4GB met SSD. Ik draai Postgres 12.5. 

```
DSMR-READER
    App / Python / Database                                                       v4.15 / v3.9.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                       1.0s / 0.5s / -h / False
    Latest telegram version read / Parser settings                                                 "42" / "4"
DATA
    Telegrams total (est.)                                                                            2214167
    Consumption records electricity / gas (est.)                                                366698 / 8463
```

Zou fijn als ik dit werkend kan krijgen. Alvast bedankt!

dennissiemensma, 2021-04-05:
Bedankt voor je melding. Er is een zeer grote kans dat dit komt door een offset in tijdzones bij de import. Ook gezien je tweede screenshot met de tijden bij de gasstanden, daar kun je goed zien dat het waarschijnlijk hetzelfde issue is.

Zie #1322 voor meer informatie en een oplossing.

tdn131, 2021-04-05:
Thanks @dennissiemensma . Ik heb het commando uitgevoerd. Vannacht zal het een update de dag- en uurstatistieken opnieuw berekenen, dus verwacht dat het morgen goed staat.

Denk je niet dat er sprake is van een bug? Het gekke is dat het goed geïmporteerd staat bij stap 1 en er vervolgens bij 2 en 3 compleet andere datumtijdvelden van maakt. Ik had verwacht dat hij dezelfde waardes uit import zou pakken.

dennissiemensma, 2021-04-05:
Het kan een bug zijn, echter is de kans groot dat het komt door externe invoer en ik kan het maar beperkt foolproof maken. Want als je screenshot klopt dan voegt het script voor elke dag twee metingen toe, waar het normaal gesproken een veel hogere dichtheid heeft. Daarmee vertrouwt het script op de incrementele verwerking in DSMR-reader, die vast een heel eind komt, maar zeker niet gebouwd is voor zulke geringe invoer.
In plaats daarvan is het waarschijnlijk robuuster als historische imports direct in `dsmr_stats_daystatistics` (en evt `dsmr_stats_hourstatistics`) gezet worden, zodat DSMR-reader er ook niet meer aan komt en ze gewoon visualiseert zoals ze ooit (extern) verwerkt zijn. Alleen zal iemand dan het script moeten wijzigen of een andere maken.

dennissiemensma, 2021-04-05:
Heb je trouwens nog in die query de datum aangepast naar je eigen overgangsdatum `10-7-2020`?

Verder hoef je denk ik niet te wachten op de nacht. Als je de instructies volgt dan kun je ze direct opnieuw laten berekenen.

tdn131, 2021-04-06:
Het probleem is opgelost! Ik had de datum inderdaad aangepast. Ik beschik niet over de knowhow om het Domoticz naar DSMR script te wijzigen. Het lijkt dat gas en elektra op dezelfde manier importeren, maar dat de verwerking op twee verschillende manieren gaat. Mogelijk dat daar verschillende timezones gebruikt worden in de backend? Thanks @dennissiemensma

tdn131, 2021-04-06:
Ik had de datum in de query aangepast. De gasmetingen van 2017 t/m juni 2020 zijn nu wel zichtbaar. Bedankt! @dennissiemensma Mogelijk dat er nog wel wat geks zit in de backend van DSMR want het lijkt er op dat hij de tijdzone op twee verschillende manieren wijzigt in de database.

dennissiemensma, 2021-04-06:
Ah fijn om te horen! Dan laat ik het voor nu voor wat het is. Ik heb in de comment, van het issue waar het script in staat, een extra waarschuwing gezet.

closed 2021-04-06 by dennissiemensma