dsmrreader/dsmr-reader #1353: Dsmr addon voor home assistant: geen updates?

gobartos, 2021-04-07:


niekniek89, 2021-04-07:
@gobartos ik denk dat je hier moet zijn voor o.a. updates:

https://community.home-assistant.io/t/dsmr-reader-add-on-for-home-assistant/279087

echter, zie ik dat 16 uur geleden de addon is bijgewerkt naar:

**2021-04-06: 0.4.1 Update to DSMR-reader v4.15.1**

gobartos, 2021-04-07:
Ja maar geen update hier in Home Assistant...

[image]

donnie-dominguez, 2021-04-07:
Home Assistant kijkt periodiek (denk eens per 24 uur) of er updates zijn. Dus het kan even duren. Wellicht helpt het om HA opnieuw te starten om een check te forceren. Maar dat is inderdaad meer iets voor de HA community

jeroenpeters1986, 2021-04-07:
Eens met wat @donnie-dominguez  en @niekniek89 aangegeven hadden, dit is iets voor de HA community.

closed 2021-04-07 by jeroenpeters1986

gobartos, 2021-04-07:
Ik wil nog graag weten of de add-on door dezelfde persoon gemaakt wordt of iets 'achteraf gefabriceerd' is.

niekniek89, 2021-04-07:
Addon wordt door iemand anders onderhouden (zie aangegeven HA community link). Helemaal bovenaan staat de maker van de addon.
[image]

dennissiemensma, 2021-04-07:
@gobartos ik kan me voorstellen dat het wat verwarrend is. Voor de duidelijkheid:

- DSMR-reader applicatie zelf: https://github.com/dsmrreader/dsmr-reader (huidig project)
- Docker-versie van DSMR-reader: https://github.com/xirixiz/dsmr-reader-docker (door xirixiz)
- HA add-on met DSMR-reader: https://github.com/sanderdw/hassio-addons (door sanderdw)