dsmrreader/dsmr-reader #1354: Live grafiek (specifiek terug levering)

niekniek89, 2021-04-07:
Goedemiddag,

Ik heb zojuist een upgrade uitgevoerd van 4.14 naar 4.15.1.
Als ik dan in het dashboard links naar "Live grafiek" ga, zie ik opeens dat het gedeelte "terug levering"  een soort van 
ondersteboven is.
Wellicht hoort dit zo (sinds de nieuwe updates), maar is nooit zo geweest.

[image]

donnie-dominguez, 2021-04-07:
Is inderdaad nieuw sinds de laatste update (v4.15.1)
[image]

dennissiemensma, 2021-04-07:
@niekniek89 dit is inderdaad zoals bedoeld. Ik kan me voorstellen dat het even wennen is.

Voor gebruikers die simultaan verbruiken en terugleveren zijn deze grafieken waarschijnlijk stukken duidelijker. De oude situatie ging er vanuit dat altijd ofwel verbruik ofwel teruglevering mogelijk was, maar dat bleek in de praktijk niet altijd zo te zijn.

Verder verduidelijkt dit ook de per-fasen-grafiek, waar in sommige situaties de Y-as 6x gebruikt werd, waardoor het er allemaal niet duidelijker werd. Nu is dat versimpeld naar tweemaal 3x.

closed 2021-04-07 by dennissiemensma