dsmrreader/dsmr-reader #1358: issue regarding pyyaml 5.4.1 and upgrade from dsmrreader 4.15.1

bottle12, 2021-04-08:
trying to update to the latest version of dsmrreader but receive the 
receive following error during deploy on a rpi 4 with 

Linux MonitorPi 5.10.17

running python 3.7.3 in virtual enviroment. 

(dsmrreader) dsmr@MonitorPi:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
4.15.1

 --- Checking Python version.
Running Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0]

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.

 --- Checking Python version.
Running Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0]

collecting pytz==2021.1
  Using cached https://www.piwheels.org/simple/pytz/pytz-2021.1-py2.py3-none-any.whl (510 kB)
Collecting pyyaml==5.4.1
  Using cached PyYAML-5.4.1.tar.gz (175 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp1u7ckhhz
         cwd: /tmp/pip-install-5j_7pomf/pyyaml_e975c43c9496412ab4e7c128e0613a06
    Complete output (39 lines):
    running dist_info
    creating /tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info
    writing /tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/top_level.txt
    writing manifest file '/tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/SOURCES.txt'
    skipping 'yaml/_yaml.c' Cython extension (up-to-date)
    reading manifest file '/tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file '/tmp/pip-modern-metadata-8t_km5_l/PyYAML.egg-info/SOURCES.txt'
    creating '/tmp/pip-modern-metadata-8t_km5_l/PyYAML.dist-info'
    Traceback (most recent call last):
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
        main()
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 140, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 126, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 295, in <module>
        python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*',
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/dist_info.py", line 36, in run
        bdist_wheel.egg2dist(egg_info.egg_info, dist_info_dir)
      File "/usr/local/lib/python3.7/dist-packages/wheel/bdist_wheel.py", line 414, in egg2dist
        shutil.copy(license, os.path.join(self.distinfo_dir, license_filename))
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/posixpath.py", line 80, in join
        a = os.fspath(a)
    TypeError: expected str, bytes or os.PathLike object, not NoneType
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz#sha256=607774cbba28732bfa802b54baa7484215f530991055bb562efbed5b2f20a45e (from https://pypi.org/simple/pyyaml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*). Command errored out with exit status 1: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp1u7ckhhz Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pyyaml==5.4.1

dennissiemensma, 2021-04-08:
Bedankt voor je vraag, lost dit het wellicht voor je op?

```
sudo su - dsmr
export SETUPTOOLS_USE_DISTUTILS=stdlib
./deploy.sh
```

bottle12, 2021-04-08:
Nee helaas

dennissiemensma, 2021-04-08:
Welke versie van `pip` draai je bij DSMR-reader?

```
sudo su - dsmr
pip --version
```

bottle12, 2021-04-08:
pip --version
pip 21.0.1 from /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/pip (python 3.7)

bottle12, 2021-04-08:
heb .cache ook leeggemaakt en opnieuw laten vullen

dennissiemensma, 2021-04-08:
Wat krijg je hierbij te zien?

```
sudo su - dsmr
pip install PyYAML --upgrade
```

bottle12, 2021-04-08:
je bent een topper :) 

(dsmrreader) dsmr@MonitorPi:~/dsmr-reader $ pip install PyYAML --upgrade
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting PyYAML
  Using cached PyYAML-5.4.1.tar.gz (175 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: PyYAML
  Building wheel for PyYAML (PEP 517) ... done
  Created wheel for PyYAML: filename=PyYAML-5.4.1-cp38-cp38-linux_armv7l.whl size=543696 sha256=c71412e9abd03e1076b9a6d2db8cc290d75205d88c3c29e72a3102c2e8798570
  Stored in directory: /home/dsmr/.cache/pip/wheels/dd/c5/1d/5d7436173d3efd4a14dcb510eb0b29525ecb6b0e41489e716e
Successfully built PyYAML
Installing collected packages: PyYAML
Successfully installed PyYAML-5.4.1

installatie loopt nu door

bottle12, 2021-04-08:
loop nu tegen database migration fout aan 

>>>> [!] Executing database migrations failed! <<<<<
       [i] Trying to automatically resolve with 'dsmr_sqlsequencereset'.
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 377, in execute
    django.setup()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mathfilters'
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 401, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 377, in execute
    django.setup()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/__init__.py", line 24, in setup
    apps.populate(settings.INSTALLED_APPS)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/registry.py", line 91, in populate
    app_config = AppConfig.create(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mathfilters'
>>>>> [!] Executing database migrations failed again! <<<<<
        - Running dsmr_sqlsequencereset did not resolve the problem.
        - Create an issue on GitHub and attach the exeception trace listed above.
[!] Halted. Post-deployment script exited with non-zero code

bottle12, 2021-04-08:
ja 

 --- Checking & synchronizing base requirements for changes.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: autopep8==1.5.6 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.5.6)
Requirement already satisfied: coreapi==2.3.3 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.3.3)
Requirement already satisfied: crcmod==1.7 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (1.7)
Requirement already satisfied: django==3.1.7 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.1.7)
Requirement already satisfied: django-admin-rangefilter==0.7.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.7.0)
Requirement already satisfied: django-admin-sortable==2.2.3 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2.3)
Requirement already satisfied: django-colorfield==0.4.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (0.4.1)
Requirement already satisfied: django-debug-toolbar==3.2 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.2)
Requirement already satisfied: django-filter==2.4.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (2.4.0)
Requirement already satisfied: django-mathfilters==1.0.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.0.0)
Requirement already satisfied: djangorestframework==3.12.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (3.12.4)
Requirement already satisfied: django-solo==1.1.5 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (1.1.5)
Requirement already satisfied: dropbox==11.5.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (11.5.0)
Requirement already satisfied: gunicorn==20.1.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (20.1.0)
Requirement already satisfied: influxdb==5.3.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (5.3.1)
Requirement already satisfied: paho-mqtt==1.5.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (1.5.1)
Requirement already satisfied: pyserial==3.5 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (3.5)
Requirement already satisfied: pyserial-asyncio==0.5 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (0.5)
Requirement already satisfied: python-dateutil==2.8.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (2.8.1)
Requirement already satisfied: python-decouple==3.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (3.4)
Requirement already satisfied: pytz==2021.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 21)) (2021.1)
Requirement already satisfied: pyyaml==5.4.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 22)) (5.4.1)
Requirement already satisfied: requests==2.25.1 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 23)) (2.25.1)
Requirement already satisfied: urllib3==1.26.4 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 24)) (1.26.4)
Requirement already satisfied: pycodestyle>=2.7.0 in /home/dsmr/.local/lib/python3.8/site-packages (from autopep8==1.5.6->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.7.0)
Requirement already satisfied: toml in /home/dsmr/.local/lib/python3.8/site-packages (from autopep8==1.5.6->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.10.2)
Requirement already satisfied: coreschema in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (0.0.4)
Requirement already satisfied: uritemplate in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (3.0.1)
Requirement already satisfied: itypes in /home/dsmr/.local/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.2.0)
Requirement already satisfied: asgiref<4,>=3.2.10 in /home/dsmr/.local/lib/python3.8/site-packages (from django==3.1.7->-r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.3.4)
[... cut ...]
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/apps/config.py", line 90, in create
    module = import_module(entry)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'mathfilters'

bottle12, 2021-04-08:
de module bestaat binnen python

Python 3.8.9 (default, Apr  7 2021, 20:02:04)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mathfilters
>>>

dennissiemensma, 2021-04-08:
Kan het zijn dat je verschillende python-versies en/of envs door elkaar heen gebruikt? 
Ik zie in je laatste comment `3.8` en in die van de deploy op `3.7`.

Los daar van blijft die fout raar, want de deploy lijkt hem wel te hebben:
```
...
Requirement already satisfied: django-mathfilters==1.0.0 in /home/dsmr/.local/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (1.0.0)
...
```

bottle12, 2021-04-08:
ik zie dat het niet bestaat in python 3.7 
heb
 python3.7 -m pip install django-mathfilters gedaan en dat resulteerde erin dat het nu wel werkt

thnx man

bottle12, 2021-04-08:
deploy.sh lijkt te kijken naar 3.7.3

>>>
(dsmrreader) dsmr@MonitorPi:~/dsmr-reader $ ./deploy.sh

 --- You are currently running version:
4.15.1

 --- Checking Python version.
Running Python 3.7.3 (default, Dec 20 2019, 18:57:59)
[GCC 8.3.0]

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Already up to date.

 >>> Running post-deployment script. <<

terwijl commando python versie 3.8.9 start

dennissiemensma, 2021-04-08:
Ik denk dat je twee versies door elkaar heen gebruikt.

De requirements lijken hier te staan:
```
/home/dsmr/.local/lib/python3.8/
```

Maar de virtualenv van DSMR-reader zelf (en dus ook de migrations):
```
/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7
```

bottle12, 2021-04-08:
wat is handig om te doen mbt die versies?

dennissiemensma, 2021-04-08:
Wellicht wil je de virtualenv helemaal verwijderen en een nieuwe maken op Python 3.8, mits die op je systeem staat (lijkt van wel).

De checkout van DSMR-reader zelf kan gewoon blijven staan.

Iets in deze trend:

```
sudo supervisorctl stop all
sudo rm -rf /home/dsmr/.virtualenvs

sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --python python3
sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt

sudo su - dsmr
./deploy.sh
logout

sudo supervisorctl start all
```

dennissiemensma, 2021-04-08:
Eventueel kun je overigens de oude virtualenv hernoemen ipv verwijderen. Die map opnieuw aanmaken is dan ook niet meer nodig.

bottle12, 2021-04-08:
Top man

Thnx voor je support en meedenken

Nog een hele fijne avond gewenst