dsmrreader/dsmr-reader #1360: Schaal terugleveringgrafiek (groene deel) gelijktrekken aan rode (verbruik) deel

AbelLykens, 2021-04-09:
Ik zou het handig vinden als de groene grafiek (teruglevering) visueel vergelijkbaar wordt met het rode deel. Als de (y) assen voor negatief en positief hetzelfde zijn, dan worden de grafieken nuttiger.

Nu zijn de rode pieken niet (eerlijk) te vergelijken met de groene teruglevering :).

[image]

dennissiemensma, 2021-04-09:
Bedankt voor je suggestie. Het lijkt er op dat ik dit niet dynamisch kan instellen, dus dan moet ik het baseren op de hoogste waarde van alle datapunten, ook degene die **buiten** de huidige zoom-selectie onder de grafiek vallen.

Opzich lijkt dat initieel een werkbare oplossing, alleen heeft dat als gevolg dat zodra er weer **lage** waarden komen, de max ervoor zorgt dat het flink vertekent. Dus als je 100 datapunten hebt rond de `500W` en eentje van `4000W`, dan blijft die `4000W` altijd de max.

[image]

----

Wellicht is het dan beter om de schaling dynamisch te houden, maar de interval tussen de Y-as-lijnen (die **nu nog** dynamisch zijn per grafiek) gelijk te trekken tussen beiden. 
Dat wordt dan een subtiele hint die zeer goed weergeeft wat de verhouding is tussen de grafieken. Als ze vergelijkbaar zijn, zoals in je eigen screenshot met daar de interval van 300 W, dan valt het amper op. 

Maar als ik ze hardcode op basis van wat magie en ze ook gelijktrek, dan zie je in de voorbeelden hieronder duidelijk de verschillen aan de hand van de Y-as-lijnen:

[image]

[image]

Wellicht is dat nog een goede verbetering, namelijk met behoud van de autoschaling.

AbelLykens, 2021-04-09:
Hm ja allebei opties... Hmm. Denk dat ik meer fan ben van de 1e oplossing, dat kijkt toch makkelijker...? ;)

dennissiemensma, 2021-04-09:
Wellicht kun je beter zien wat ik bedoel met dit voorbeeld. Voor het gemak aan testdata heb ik hem even opgeblazen naar 35k W tov 1k W, maar doe het gerust een factor 10 minder (3500 W tov 100 W), de verhoudingen blijven immers gelijk voor waar ik op doel.

Wat je hier ziet is dat **ergens** op de tijdlijn een waarde van 35k is geweest, maar die is buiten de huidige zoom. Dat kan dus een waarde zijn van gisteren of zelfs 6 dagen terug, afhankelijk van de instellingen van de gebruiker. Omdat ik het niet dynamisch kan maken, heeft het voorstel invloed op de max voor de rest van de data. 

[image]

En dat is precies waar ik de uitdaging in zie. Op het moment dat de hoogste waarde niet meer in beeld is, blijft de max Y wel de rest beinvloeden en ik vind het absoluut niet duidelijk. 
Het is ook een van de redenen dat ik deze grafieken ooit heb gekozen, omdat ze supergoed automatisch schalen op basis van de zoom onder de grafiek. 

De enige situatie waarin dit **wel** zou werken, is wanneer de complete tijdlijn in de zoom staat, waardoor die piek (die in dit screenshot niet zichtbaar is) dan wel erbij staat en dan kloppen de onderlinge verhoudingen wel weer.

----

Ik ben er nog niet over uit wat dan wel de oplossing is. Ik zal nog wel een keertje wat spelen met of ik iig zowel de autoschaling als de Y-as-lijnen samen werkend krijg voor beide assen.

AbelLykens, 2021-04-10:
Zo wordt het een stuk duidelijker. Niet ideaal, dat ben ik met je eens!

Ben benieuwd of "nog een keertje spelen" nieuwe inzichten brengt. 

"Probleem" is in ieder geval duidelijk ;)

donnie-dominguez, 2021-04-13:
Hmm, ik wilde eigenlijk een zelfde soort vraag gaan stellen, maar met deze inzichten verandert de zaak denk ik inderdaad. Een fase in mijn meterkast kan maximaal 25A leveren (er zijn ook meterkasten waar dit 40A is volgens mij). In mijn geval zou er dus per fase max 5750W geleverd kunnen, en dat keer 3. Ik mag hopen van niet, maar in theorie kan er dus een waarde van ruim 15.000W in de grafiek staan. Als dit een kortstondige piek is, en daarna kruip ik weer terug naar mijn huidige 'sluipverbruik' van tussen de 100 en 250W dan is dat zo goed als onzichtbaar in de grafiek. Dat lijkt mij ook niet de mooiste oplossing. Fijn dat er goed over na wordt gedacht voordat er veranderingen/verbeteringen worden doorgevoerd.

cmderks93, 2021-04-15:
Ik reageer ook even, meer omdat ik op de hoogte gehouden wil worden :-p
Maar de vraagstelling heb inderdaad ook.

pa1pdr, 2021-06-10:
Ik ben ook voorstander van een lineaire y-as (ongebroken bij 0), maar geen voorstander van negatief tonen van teruglevering: hierdoor nemen de grafieken de dubbele 'screen estate' in. Kan het negatief tonen van terugleveren een optie worden? Dat lost voor mij tegelijkertijd het y-as probleem op (er is nl geen negatieve meer ;-))

dennissiemensma, 2021-06-10:
@pa1pdr bedankt voor je aanvulling, ik begrijp je punt. Helaas is het voor iedereen een eigen voorkeur en het is onwerkbaar om tig varianten en combinaties hiervoor te ondersteunen. Ik weet dan ook niet of de huidige variant uiteindelijk de definitieve versie blijft, maar daarvoor zal eerst de schaling rechtgetrokken moeten worden om iedereen een goed oordeel te laten vellen.

Wat betreft de grootte van de grafieken, het is onbegonnen werk om daar iets te pakken wat voor iedereen werkt. Want er zijn zoveel verschillende schermgroottes, zowel qua monitoren als telefoons, om nog maar te zwijgen over 4k of retina, die elk weer hun eigen ervaring en proporties van de grafieken tonen. 
En de een wil meerdere grafieken in 1 oogopslag zien en de andere juist weer 1 grafiek in het groot. Het zal altijd ergens een compromis in het midden worden helaas.

D1StrX, 2022-11-25:
Weet niet of het volgende hierop aansluit;
Maar nu laat dsmr 0 (live) verbruik zien wanneer de zonnepanelen meer opwekken dan gebruikt wordt.
Kan middels de of een grafiek nog steeds het actuele verbruik weergeven worden? Of is dit helaas de werking van de slimme meter. Potentieel met een aftreksom.

dennissiemensma, 2022-11-25:
Je kunt het beste even een nieuw issue maken met wat meer informatie of voorbeelden. Dit huidige issue is al bijna 2 jaar oud.

dennissiemensma, 2023-10-23:
Ik ben even kwijt wat de follow-up hiervoor was. Als ik het zo teruglees dan was de huidige opzet een probeersel, maar is dat probeersel inmiddels al redelijk lang in productie.

closed 2024-01-26 by dennissiemensma