dsmrreader/dsmr-reader #1361: MQTT

DavyRoswinkel, 2021-04-11:
Would like to see a possibility to receive the telegrams by MQTT instead of telnet telegrams.  
I use a (p1) gateway that has the possibility to send telnet telegrams or send MQTT telegrams. MQTT telegrams are more stable but not usable in the DSMR-reader at this moment.

dennissiemensma, 2021-04-12:
Bedankt voor je verzoek! Een van de redenen dat dit niet ondersteund wordt is omdat er geen MQTT-standaard is voor DSMR over MQTT.
Ik heb al diverse varianten gezien van ruwe telegrammen tot aan JSON en split topic. En daarbij waren ook nog eens niet altijd alle velden aanwezig. Plus dat dat betekent dat voor ruwe telegrammen de checksum-validatie ook nog eens gedaan moet worden en voor de andere weer niet. 

Vandaar dat de huidige wegen de ingebouwde of remote datalogger zijn. Eventueel met een API als alternatief waarbij ofwel het ruwe telegram gestuurd kan worden, ofwel het geparste telegram (checksum-validatie wordt dan niet gedaan).

DavyRoswinkel, 2021-04-12:
Dank je voor de heldere uitleg. Wel jammer. De gateway die ik heb geeft op 2 manieren de gegevens door, via telnet en via MQTT. Via telnet worden de gegevens verwerkt door de dsmrreader, een prachtige mooie overzichtelijke addon. Echter het is minder stabiel en veel belastender als de MQTT verwerking.( ligt niet aan de addon.) Het is mij niet bekend dat er gegevens zouden ontbreken in de MQTT gegevens. Misschien dus voor de mensen die zich daarmee bezig houden om daar eens overeenstemming in te gaan vinden en uniformiteit aan te brengen. Bedankt voor nu.

dennissiemensma, 2021-04-13:
Ik denk dat het hetzelfde is als met API's. Iedereen maakt net weer zn eigen voorkeur of variant, voor wat diegene zelf denkt dat nodig is. 
Opzich is dat prima en zo hangt de halve wereld aan elkaar geknoopt, alleen zit de vraag vaak meer in bij wie het werk komt te leggen. Ik kan niet verlangen dat elke ontwikkelaar alles op een manier doet en ook nog een manier dat DSMR-reader het beste uitkomt. Anderzijds kan ik ook niet elke mogelijke variant ondersteunen en blijven onderhouden.

Dus uiteindelijk zal het iets er tussenin worden, vermoedelijk in de vorm van een pluginscript waar men her en der zelf dan maar de eigen situatie moet bijstellen. Het is een alternatief op een script aan de andere zijde, die de API van DSMR-reader gebruikt.

closed 2021-04-29 by DavyRoswinkel