dsmrreader/dsmr-reader #1362: kosten op basis van selectie grafiek

sowen222, 2021-04-13:
Ik zou graag een selectie in de 'live' grafiek willen maken, en de totale verbruik+kosten van die selectie op basis van mijn huidige contract willen kunnen zien.

Dus in bijvoorbeeld de 'Recent electricity usage' grafiek, mijn muis gebruiken om een blok te selecteren, en gelijk in een popup totaal verbruik + kosten voor die selectie zien.

Dat maakt het direct inzichtelijk wat bepaalde activiteiten kosten. Bijvoorbeeld, als ik de ovenreiniging aanzet, en die 2u20m draait - voor een deel in hoog tarief en een deel in laag tarief - zou het mooi zijn als ik in de grafiek die periode van 2u20m kon selecteren, en de verbruik+kosten automatisch voor mij berekend werden.

dennissiemensma, 2021-04-13:
Bedankt voor je verzoek. Ik weet niet in hoeverre dit mogelijk is, ook omdat er behoorlijk wat complexiteit komt kijken in zowel de grafieken als de prijsberekening. Desnietemin zal ik er in de toekomst een keer wat dieper naar kijken of er iets mogelijk is.

erikkral, 2021-04-14:
Ik heb dezelfde wens. Ik was vanochtend benieuwd hoeveel mijn vaatwasser verbruikt die ik ‘s nachts laat draaien. Nu is de live grafiek niet de beste manier om het te meten omdat er ook andere apparaten aan staan maar het geeft me een enig beeld omdat de vaatwasser de enige grootverbruiker is.

Als je op basis van een selectie in de live grafiek al een indicatie krijgt van verstookte kWh’s, zou dat al fijn zijn. 

Voor wat betreft de tarieven, misschien ligt het aan mij, maar de maandfactuur van greenchoice komt bij mij niet overeen met het maandbedrag wat DSMR Reader aangeeft. Mijn beeld is dat dat veroorzaakt wordt doordat je in DSMR reader alleen de KWh-prijs kan invoeren echter de factuur bestaat ook nog uit een andere kosten. Of ik doe iets verkeerds....

sowen222, 2021-04-14:
@erikkral - ik gebruik DSMR reader pas maar net, maar volgens mij zal er een verschil in maandbedrag komen door saldering. Dus als je zonnepanelen hebt, lever je terug tegen hetzelfde tarief als het tarief voor afnemen, en alleen het overschot aan wat je teruglevert is tegen een lager 'teruglevertarief'. En (anders) kijk ook naar de dagelijkse vaste kosten - dit moet alle vaste kosten opgeteld zijn (dus vaste levering + netbeheer + belasting(vermindering), van zowel gas als stroom.

dennissiemensma, 2021-04-14:
Voor jullie informatie ter aanvulling op de laatste twee comments: DSMR-reader heeft niet als doel **om de prijsberekening van je energieleverancier** te spiegelen. Het is onmogelijk omdat goed te krijgen voor elke leverancier en is daarom bewust zo 'simpel' gemaakt in DSMR-reader. 

Alle prijzen in DSMR-reader zijn bedoeld om jullie een gevoel te geven wat "een dag kost" en dan vooral in vergelijking met andere dagen, waarbij het seizoen of teruglevering een factor kan zijn. En bijvoorbeeld hoeveel het kost om de hele dag de verwarming aan te zetten, bijvoorbeeld zelf schrok ik daar in het begin namelijk van, dus ik weet nu goed wat ik kan verwachten.

_Als jullie behoefte hebben om "iets" te vergelijken bij je leverancier, dan zou mijn advies zijn om naar de meterstanden op de afrekening te kijken._ 
_Zelf doe ik dat wanneer ik wissel van leverancier, ook om te checken of wel de goede peildatum is gebruikt. Tot nu toe kwam het bij mij nog telkens overeen met wat in DSMR-reader staat, hooguit 1 kWh verschil._

dennissiemensma, 2021-04-18:
Dit issue is dubbel met #979 zie ik nu, wat dezelfde wens is, maar die niet meer op de voorpagina met issues staat. 

Ik sluit even de huidige, het beste is om te subscriben op #979.

closed 2021-04-18 by dennissiemensma