dsmrreader/dsmr-reader #1364: summary page, totalen voor afgelopen 365 dagen

sowen222, 2021-04-14:
Ivm saldering in NL en het feit dat ik zeker overproductie (+/- 1000 kWu op jaarbasis) heb, heb ik mijn levertarieven gelijk aan mijn teruglever tarieven (dus alle 4 staan nu op 0,08). Wat ik oplever streept wat ik verbruik weg, en de overschot is dus wat ik werkelijk verdien.

Maar om het nóg inzichtelijker te maken, zou het handig zijn als op de 'summary page' er - naast dag/maand/jaar - ook een tabel voor de afgelopen 365 dagen was. Zo kan ik aan het einde van een contractjaar zien wat ik +/- op mijn jaarrekening zou moeten staan. Als dat erg ingewikkeld is, dan in ieder geval een tabel voor de afgelopen 12 maanden.

dennissiemensma, 2021-04-14:
Bedankt voor je verzoek. Ik denk dat je misschien al geholpen bent met de CSV-export feature van de dagtotalen. 

Deze functionaliteit is wat ondergesneeuw tussen alle andere opties, maar je kunt er iig bij als je op "CSV" zoek in het configuratie-tussenscherm, of door direct naar deze link te gaan in DSMR-reader:
```
/export

# Bijvoorbeeld:
http://ip-adres-van-je-pi/export
```

dennissiemensma, 2021-04-14:
Ik zie trouwens dat de "vaste kosten" nog missen in die export. Ik zal die kolom toevoegen bij een volgende release, maar desnietemin denk ik dat je een heel eind komt met de gegevens uit de export, op dagniveau.

dennissiemensma, 2021-04-18:
De kolom "vaste kosten" is nu ook toegevoegd in [v4.15.2](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.15.2)

closed 2021-04-18 by dennissiemensma