dsmrreader/dsmr-reader #1366: Gemiddelde spanning.

cmderks93, 2021-04-15:
De grafiek van de spanning per fase bekijk ik met enige regelmaat. Naar mijn mening zou het een mooie toevoeging zijn om per fase de gemiddelde spanning over een bepaalde periode te kunnen bekijken. 

Wellicht dat het huidige geselecteerde grafiekje gebruikt kan worden voor de input waardes.

dennissiemensma, 2021-04-15:
Bedankt voor je verzoek. Ik denk dat daar wel wat voor te maken is. 

Ik weet alleen niet op welke termijn. Als ik het via de grafieken-library zelf kan doen, scheelt het tijd. Anders is het een wat grotere wijziging (met relatief lage prio).

dennissiemensma, 2021-09-19:
Ik verwacht niet dat dit op korte of middellange termijn opgenomen kan worden. 

Voor nu ben je wellicht het beste geholpen met een query die je zelf kunt uitvoeren:

```
sudo su - postgres
psql dsmrreader

select count(id) as "readings", avg(phase_voltage_l1) "avg-l1", avg(phase_voltage_l2) "avg-l2", avg(phase_voltage_l3) "avg-l3" from dsmr_datalogger_dsmrreading where timestamp between '2021-09-01 00:00:00+01' and '2021-09-02 00:00:00+01' limit 1;
```

- Dit geeft de gemiddeld spanning tussen 1 september middernacht en de volgende middernacht.
- Let wel dat dit alleen werkt als je de data nog hebt. DSMR-reader schoont na X periode op, dus check even waar de retentie op staat. Het aantal metingen wat meegenomen is in de query staat ook in het resultaat.

Het is uiteraard niet zo fijn als een grafiek, maar dan kun je iig meer afleiden dan nu.

dennissiemensma, 2022-07-05:
Ik laat deze voor wat het is. Ik denk dat er teveel haken en ogen aan zitten, ook omdat de input voor de grafiek maar beperkt is. Wellicht kom ik er ooit op terug bij een compleet redesign van alle grafieken.

closed 2022-07-05 by dennissiemensma