dsmrreader/dsmr-reader #1368: Handmatig meterstanden invoeren

alexswart, 2021-04-17:
Helaas ben ik een backup van dsmrreader kwijt geraakt maar heb nog wel een aantal meterstanden can de energieleverancier op specifieke datums kan ik deze ook handmatig invoeren om zo toch de jaarverbruiken te kunnen zien

dennissiemensma, 2021-04-17:
Bedankt voor je vraag. DSMR-reader heeft helaas geen optie die dit makkelijk voor je maakt. Dat zeggende hebben, afhankelijk van om hoeveel metingen/jaren het gaat kun je er wel een slimmigheidje voor gebruiken. 

Uiteindelijk gebruiken de pagina's van het Archief en Energiecontracten alle dagtotalen die binnen een periode vallen (maand of jaar of contractduur) en tellen die op voor de weergave.
Dat betekent dat je eventueel per jaar of contractduur gewoon **1 dag** kan invullen met **al het verbruik** van die periode. Voor een jaar of contractperiode heb je dan 1 dag waar alles in zit. Eventueel kun je dat zelfs per maand doen, als je die gegevens hebt en daar de tijd in wilt steken, maar dat kost je dus wel 12x zoveel handmatige invoer. 

Je zult hiermee wel zelf het verschil (dus verbruik) moeten berekenen, want DSMR-reader doet dat alleen vanaf meting-niveau en niet dag-niveau, dus dat betekent wel iets rekenwerk. Vandaar ook de keuze die je zelf hebt om te bepalen hoeveel losse dagen je in jouw geval wilt invullen.

Je kunt hier een datum met gegevens invoeren en kijken hoe het er uit komt voor je:
```
/admin/dsmr_stats/daystatistics/add/
```

dennissiemensma, 2021-04-22:
@alexswart is het gelukt?

closed 2021-04-26 by dennissiemensma