dsmrreader/dsmr-reader #1369: Feature: Multiple DSMR's versions on a single PostgreSQL server.

remyderuysscher, 2021-04-18:
Apparently DSMR-reader is using pg_default to store django_admin_log_pkey. When the second instance on the database server is used to change the settings of the MQTT server this error is displayed:

```
Server Error
Sorry, something unexpected happened.
Exception:

IntegrityError: duplicate key value violates unique constraint "django_admin_log_pkey"
DETAIL:  Key (id)=(29) already exists.

Traceback:

  File "/usr/local/lib/python3.9/site-packages/django/core/handlers/base.py", line 181, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/sites.py", line 233, in inner
    return view(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/solo/admin.py", line 74, in change_view
    return super(SingletonModelAdmin, self).change_view(

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1656, in change_view
    return self.changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 43, in _wrapper
    return bound_method(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/utils/decorators.py", line 130, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1534, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 1587, in _changeform_view
    self.log_change(request, new_object, change_message)

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/options.py", line 826, in log_change
    return LogEntry.objects.log_action(

  File "/usr/local/lib/python3.9/site-packages/django/contrib/admin/models.py", line 29, in log_action
    return self.model.objects.create(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 447, in create
    obj.save(force_insert=True, using=self.db)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/base.py", line 753, in save
    self.save_base(using=using, force_insert=force_insert,

  File "/usr/local/lib/python3.9/site-packages/django/db/models/base.py", line 790, in save_base
    updated = self._save_table(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/base.py", line 895, in _save_table
    results = self._do_insert(cls._base_manager, using, fields, returning_fields, raw)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/base.py", line 933, in _do_insert
    return manager._insert(

  File "/usr/local/lib/python3.9/site-packages/django/db/models/manager.py", line 85, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 1254, in _insert
    return query.get_compiler(using=using).execute_sql(returning_fields)

  File "/usr/local/lib/python3.9/site-packages/django/db/models/sql/compiler.py", line 1397, in execute_sql
    cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 66, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 75, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/usr/local/lib/python3.9/site-packages/django/db/utils.py", line 90, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/usr/local/lib/python3.9/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

```

dennissiemensma, 2021-04-18:
Bedankt voor je melding. Ik heb wat meer context van je nodig, vooral hoe je het ingericht hebt qua databases. Ik zou wel verwachten dat het mogelijk is om de databases van twee instanties op dezelfde server te hosten. Een lokale test geeft hier dan ook geen issues.

Hebben beide DSMR-reader instanties een aparte **database** op de PostgreSQL-server? 

Mocht je het namelijk anders ingericht hebben, door bijvoorbeeld verschillende **schema's** binnen 1 DB, dan zul je meer in deze richting moeten kijken: #1138

remyderuysscher, 2021-04-18:
Zeker beide instanties hebben een aparte database en aparte credentials, deze wordt ook zo in docker gebruikt. Ik heb niets veranderd in PostgeSQL server (standaard docker container versie 13). De bug is 100% reproduceerdbaar. Ik kan nog een test doen met backup- en restore database en ook het verwijderen van de container.

dennissiemensma, 2021-04-18:
Het kan ook zijn dat het een individueel issue is in de database zelf. Je kunt kijken of dit het wellicht voor je oplost:

https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/error-key-already-exists.html

remyderuysscher, 2021-04-18:
> 
> 
> Het kan ook zijn dat het een individueel issue is in de database zelf. Je kunt kijken of dit het wellicht voor je oplost:
> 
> https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/error-key-already-exists.html

Helaas heeft niet geholpen. Maar waarom gebruikt de instance de pg_default database en niet de database van de instance? (Ik heb verder niet gekeken in de code).

dennissiemensma, 2021-04-18:
Bedankt voor de update. DSMR-reader is gebouwd op Django en doet niets specifieks met de database-instellingen, dus ook niet bewust met een `pg_default` (inhoudelijk zegt het mij ook niets). 
Als ik lokaal in mijn beide instances iets aanpas in de admin-interface, dan zie ik in beide database gewoon losse entries in `django_admin_log` en die lijken geen last van elkaar te hebben, dus dan wordt het wat lastig reproduceren voor mij en zoeken naar een oplossing. Wellicht is er iets anders in PostgreSQL 13, die gebruik ik zelf nog niet, maar dat kan ik me bijna niet voorstellen.

Werkt je tweede instance verder wel qua dataverwerking of loopt die daar ook op stuk bij andere tabellen?

dennissiemensma, 2021-04-19:
Ben je nog wat wijzer geworden of het ook bij andere tabellen speelt?

remyderuysscher, 2021-04-19:
Helaas niet, de twee instances werken wel afzonderlijk van elkaar. Telegrammen en grafieken worden goed weergegeven. Alleen bij het opslaan wijzigingen in de admin omgeving gaat het fout. Ik zat eerder op niet meer ondersteunde versie van PostgreSQL. Daar geen problemen gehad. Verse docker en nieuwe db server opgetuigd en database restore zoals jij hebt beschreven. Ik zie wel dat beide databases dezelfde tablespace gebruiken (pg_default). Als ik deze aanpas werkt het ook nog niet. 

Waar kan ik deze key vinden? django_admin_log_pkey

dennissiemensma, 2021-04-19:
Bij nader inzien zie ik hem er wel tussen staan, vreemd dat dat het dan niet opgelost heeft toen, wellicht werkt het dan toch niet helemaal lekker:

```
BEGIN;
SELECT setval(pg_get_serial_sequence('"django_admin_log"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_admin_log";
SELECT setval(pg_get_serial_sequence('"auth_permission"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_permission";
SELECT setval(pg_get_serial_sequence('"auth_group_permissions"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_group_permissions";
SELECT setval(pg_get_serial_sequence('"auth_group"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_group";
SELECT setval(pg_get_serial_sequence('"auth_user_groups"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user_groups";
SELECT setval(pg_get_serial_sequence('"auth_user_user_permissions"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user_user_permissions";
SELECT setval(pg_get_serial_sequence('"auth_user"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "auth_user";
SELECT setval(pg_get_serial_sequence('"django_content_type"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_content_type";
SELECT setval(pg_get_serial_sequence('"dsmr_api_apisettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_api_apisettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_backendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_backendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_emailsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_emailsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_scheduledprocess"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_scheduledprocess";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_backupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_backupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_dropboxsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_dropboxsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_emailbackupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_emailbackupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_energysupplierprice"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_energysupplierprice";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_electricityconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_electricityconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_gasconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_gasconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_consumptionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_consumptionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dataloggersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dataloggersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_retentionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_retentionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dsmrreading"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dsmrreading";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_meterstatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_meterstatistics";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_meterstatisticschange"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_meterstatisticschange";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_notification"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_notification";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_frontendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_frontendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_sortedgraph"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_sortedgraph";
SELECT setval(pg_get_serial_sequence('"dsmr_mindergas_mindergassettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mindergas_mindergassettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_mqttbrokersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_mqttbrokersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsondaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsondaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_splittopicdaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_splittopicdaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_rawtelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_rawtelegrammqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsontelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsontelegrammqttsettings";
[... cut ...]
SELECT setval(pg_get_serial_sequence('"dsmr_influxdb_influxdbintegrationsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_influxdb_influxdbintegrationsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_influxdb_influxdbmeasurement"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_influxdb_influxdbmeasurement";
COMMIT;

        BEGIN;
        SELECT setval(
            pg_get_serial_sequence('"django_migrations"','id'), coalesce(max("id"), 1), max("id") IS NOT null
        ) FROM "django_migrations";
        COMMIT;
```

Hoe dan ook fijn dat de oorzaak bekend is, voor als iemand er in de toekomst weer tegenaan loopt.