dsmrreader/dsmr-reader #1373: Day statistics are lagging behind   (2 days, 20 hours ago)

jeroenvdpoll, 2021-04-25:
Ik heb 3 DSMR instanties draaien die allen dezelfde foutmelding hebben: Day statistics are lagging behind   (2 days, 20 hours ago)

De 3 instanties zijn:
Versie 4.9 op een raspberry pi
versie 4.15.1 op docker ==> eigen postgres docker met laatste versie 13.2
versie 4.15.2 op docker ==> eigen postgres docker met laatste versie 13.2

Het zijn dus 3 van elkaar los staande versies maar wel op exact hetzelfde moment een probleem hebben gehad.

Het probleem lijkt in de gas consumptie te zitten, want mindergas kan al drie dagen niet geupdate worden omdat er geen gasconsumptie zou zijn. In de dag grafiek is er wel gasconsumptie te zien overigens.

Ik kan de volgende error vinden in back-end.log:
Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 67, in export
    last_gas_reading = GasConsumption.objects.filter(
  File "/usr/local/lib/python3.9/site-packages/django/db/models/query.py", line 325, in __getitem__
    return qs._result_cache[0]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/dsmr/dsmr_mindergas/services.py", line 33, in run
    export()
  File "/dsmr/dsmr_mindergas/services.py", line 72, in export
    raise AssertionError(_('No recent gas reading found'))
AssertionError: No recent gas reading found

Wie kan me helpen?

jeroenvdpoll, 2021-04-25:
```

DSMR-READER
    App / Python / Database                                                        v4.9 / v3.7.3 / postgresql
    Backend sleep / Datalogger sleep / Retention cleanup                                   1.0s / 5.0s / 672h
    Telegram latest version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                             242114
    Consumption records electricity / gas (est.)                                                 32434 / 9505

UNRESOLVED ISSUES
    Day statistics are lagging behind                                                    2 days, 20 hours ago

```

dennissiemensma, 2021-04-25:
Bedankt voor je melding. Ik denk dat je in dit geval het beste DEBUG-logging tijdelijk aan kan zetten op die van je Pi: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

Na reload zou je in de backend-log exact moeten zien wat die doet en ook waarom die iets niet doet. Je bent dan doorgaans op zoek naar een van deze meldingen, die aangeeft waarom de dagstatistieken (nog) niet gegenereerd kunnen worden.

Die beginnen met `Stats: ...`:
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L61
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L70
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L79
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L86
- https://github.com/dsmrreader/dsmr-reader/blob/v4/dsmr_stats/services.py#L98

dennissiemensma, 2021-04-25:
En totaal ongerelateerd, _mocht_ je vanaf je Pi MinderGas-uploads doen, vergeet niet bij te werken naar v4.11 (#1253), want ik weet niet precies hoelang zij de oude API-URL in stand houden.

jeroenvdpoll, 2021-04-25:
Merci.

Ik heb de debug log aangezet op de pi. Nu eens de log doorspitten

dennissiemensma, 2021-04-25:
Je kunt eventueel de backend-sleep tijdelijk wat ophogen naar 10 of 30 seconden, zodat de log normaal te volgen is.

jeroenvdpoll, 2021-04-25:
Welke log file moet ik hebben?
In de backend kan ik geen stats vinden
In de datalogger staat alleen deze melding: Current logging level set to "ERROR".........

dennissiemensma, 2021-04-25:
Die van de `dsmr_backend`. Het kan even duren voordat die weer probeert de dagstatistieken te doen. 

In de admin op `/admin/dsmr_backend/scheduledprocess/` kun je zien wanneer `Generate day and hour statistics` weer gaat draaien. Eventueel kun je de tijd in het verleden zetten, zodat die de volgende run draait en logt.

jeroenvdpoll, 2021-04-26:
Dennis,

Vreemd, maar nu zijn alle drie de instanties weer helemaal bijgewerkt. Mindergas is ook met de 25e bijgewerkt. Ik zie in de log nu geen bijzondere foutmeldingen.

Geen idee wat er verandert is. Het enige wat ik gedaan heb is:
Raspberrypi de log level aangepast en reboot
Ene docker versie 4.15.1 ==> 14.15.2 en Postgres op nieuwste versie gezet. Dus feitelijk reboot van dsmrreader en postgress
Andere docker alleen een reboot van het hele systeem

Bedenk me dat ik gisteren ook nog de P1 logger (connexxit) opnieuw heb opgestart 

Ik zie nu dat vanaf 18 april de statistieken niet kloppen (archive view):
18e: Geen statistieken
19e: 1x electriciteit en 1x Gas
20e: geen
21e: geen
22e: 1x electriciteit
23e: geen
24e: geen
25e: normaal

Blijkbaar is er de 18e ergens iets foutgegaan. Vreemd dat ze alledrie hetzelfde resultaat geven. Zou kunnen dat de P1 logger de 18e een rare telegram heeft gestuurd waar dsmrreader niet mee om kon gaan.

Thanks voor de support. Ik hou het de komende dagen in de gaten!

dennissiemensma, 2021-04-26:
Je zou kunnen kijken bij de metingen in de admin onder `admin/dsmr_datalogger/dsmrreading/`.

Wellicht zie je daar een gat. Zo ja, dan verwacht ik dat je wellicht issues hebt gehad met de datalogger, al geeft dat niet de exacte oorzaak. Rare telegrammen worden als het goed is gewoon geweigerd en hebben geen invloed, mits DSMR-reader ze zelf parst.

neverstoppedaskingwhy, 2021-05-02:
Misschien toevallig, maar wel apart. Hier ook op een raspberry pi 4 met 4.15.2 dezelfde foutmelding. In admin/dsmr_datalogger/dsmrreading/ is de laatste waarde van 25.04.2021 4:31 am, daarna is er blijkbaar een gat in de data. Opgevallen is het omdat ook de actuele waardes vandaag niet meer werden getoond. Er was ook een waarschuwing voor de database grootte van 1097 MB. Na nogmaals een ./deploy.sh uitgevoerd te hebben, (alleen de readme.md was nieuwer),  werd er een reload gedaan en zijn de actuele waardes ook weer op orde.

dennissiemensma, 2021-05-02:
@neverstoppedaskingwhy bedankt voor je aanvulling. De melding is een constatering dat er gegevens ontbreken, maar de inhoudelijke oorzaak kan anders zijn.

Je zou nog kunnen zoeken in de logs of je daar foutmeldingen kan vinden van rond de 25e.

jeroenvdpoll, 2021-05-02:
Ik denk dat ik de oorzaak heb achterhaald.

Ik had in totaal 4 apps die de P1 reader (Connectix) tegelijk uitlazen. 3x dsmrreader en nog een appje. Ik denk dat daarom de P1 reader af en toe in de error schoot en er geen readings meer doorkwamen. Teveel simultane  telnetsessies kan die blijkbaar niet tegen.

De oorzaak lijkt niet in dsmrreader zelf te liggen. Nu draait de boel wel stabiel. Kreeg nog wel foutmeldingingen dat mindergas niet kon worden bijgwerkt, maar die zijn inmiddels ook weg. Zal wel een achterstand zijn die weggewerkt moest worden.

Thanks voor de hulp.

neverstoppedaskingwhy, 2021-05-02:
Er zijn inderdaad fouten in de logs te vinden

In de supervisord.log INFO level meldingen vanaf 25.04.2021:

2021-04-25 04:05:11,037 INFO spawned: 'dsmr_datalogger' with pid 15252
2021-04-25 04:05:12,042 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)

In dsmr_datalogger-stderr---supervisor-tGS1pA.log.3 (en .log.2, .log.1 en .log) volgende melding, die zich dan herhaalt (alle 15 - 75 sekonden):

2021-04-25 04:05:08,696 | ERROR    mixins.run_once:101 | dsmr_datalogger.management.commands.dsmr_datalogger: [!] Exception raised. Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.InternalError_: right sibling's left-link doesn't match: block 14681 links to 0 instead of expected 1 in index "dsmr_datalogger_dsmrreading_processed_add8a852"

block 14681 en processed_add8a852 is hetzelfde in iedere melding

dennissiemensma, 2021-05-02:
@jeroenvdpoll mooi, fijn dat je het gevonden hebt!

dennissiemensma, 2021-05-02:
@neverstoppedaskingwhy 
> psycopg2.errors.InternalError: right sibling's left-link doesn't match: block 14681 links to 0 instead of expected 1 in index "dsmr_datalogger_dsmrreading_processed_add8a852"

Dat zijn geen fijne meldingen, gezien het uit de database intern lijkt te komen. Ik weet niet hoe/of je het kunt oplossen. Ik lees her en der wat over corruptie en dat **kan** (is dus niet zeker!) door corruptie in de opslag komen.

Ik zou hoe dan ook per direct een database backup maken en veiligstellen buiten je SD-kaart. Let wel dat een backup alsnog intern corrupt kan zijn, dus en probeer vooral ook via DSMR-reader op `/export` een CSV export te maken van alle dagstatistieken. En scroll even door die CSV of je iets geks ziet. Bewaar hem sowieso.

Het probleem met corruptie is dat het nooit zichtbaar is tenzij het te laat is. En het is lastig te detecteren. Want het kan dus ook gewoon een systeemfout zijn en dat er niets met je SD-kaart aan de hand is.

closed 2021-05-02 by dennissiemensma