dsmrreader/dsmr-reader #1374: Upgrade Error bij Collecting pyyaml==5.4.1

jossie67, 2021-04-26:
Hi

Ik probeer mijn huidige versie DSMR-reader v4.13. te upgraden naar de laatste versie.
Voor een upgrade doe ik altijd een apt update en een apt upgrade
Ik draai onder Ubuntu 18.04 op een Proxmox server, ik maak altijd eerst een backup.

Dit is het begin van de terminal sessie.

`
_(dsmrreader) dsmr@dsmr:~/dsmr-reader$ ./deploy.sh 

 --- You are currently running version: 
4.13

 --- Checking Python version.
Running Python 3.6.9 (default, Jul 17 2020, 12:50:27) 
[GCC 8.4.0]
[WARNING] Python version is deprecated. Preferred version for DSMR-reader is 3.7.0 or higher.

 --- Checking for local file changes.

 --- Pulling remote repository for new commits...
remote: Enumerating objects: 2193, done.
remote: Counting objects: 100% (1719/1719), done.
remote: Compressing objects: 100% (494/494), done.
remote: Total 2193 (delta 1340), reused 1497 (delta 1145), pack-reused 474
Receiving objects: 100% (2193/2193), 3.02 MiB | 2.91 MiB/s, done.
Resolving deltas: 100% (1570/1570), completed with 169 local objects.
From https://github.com/dsmrreader/dsmr-reader
   82fb9bce..2259101b  v4          -> origin/v4
   70e22df3..2259101b  development -> origin/development
 * [new branch]        poc/development/echarts-5 -> origin/poc/development/echarts-5
 * [new tag]           v4.14.0     -> v4.14.0
 * [new tag]           v4.15.0     -> v4.15.0
 * [new tag]           v4.15.1     -> v4.15.1
 * [new tag]           v4.15.2     -> v4.15.2

Collecting python-decouple==3.4
  Using cached python_decouple-3.4-py3-none-any.whl (9.5 kB)
Collecting pytz==2021.1
  Using cached pytz-2021.1-py2.py3-none-any.whl (510 kB)
Collecting pyyaml==5.4.1
  Using cached PyYAML-5.4.1.tar.gz (175 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpvvbbpfl9
         cwd: /tmp/pip-install-8ij57en7/pyyaml_f0e84f9a0f0249eb87603f897078f682
    Complete output (39 lines):
    running dist_info
    creating /tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info
    writing /tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/PKG-INFO
    writing dependency_links to /tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/dependency_links.txt
    writing top-level names to /tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/top_level.txt
    writing manifest file '/tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/SOURCES.txt'
    skipping 'yaml/_yaml.c' Cython extension (up-to-date)
    reading manifest file '/tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file '/tmp/pip-modern-metadata-7d7nckl9/PyYAML.egg-info/SOURCES.txt'
    creating '/tmp/pip-modern-metadata-7d7nckl9/PyYAML.dist-info'
    Traceback (most recent call last):
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 280, in <module>
        main()
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 113, in prepare_metadata_for_build_wheel
        _run_setup()
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 72, in _run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 295, in <module>
        python_requires='>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*',
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 129, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
        dist.run_commands()
      File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
        self.run_command(cmd)
      File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
        cmd_obj.run()
      File "/usr/lib/python3/dist-packages/setuptools/command/dist_info.py", line 36, in run
        bdist_wheel.egg2dist(egg_info.egg_info, dist_info_dir)
      File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 443, in egg2dist
        shutil.copy(license, os.path.join(self.distinfo_dir, license_filename))
      File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/posixpath.py", line 80, in join
        a = os.fspath(a)
    TypeError: expected str, bytes or os.PathLike object, not NoneType
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/a0/a4/d63f2d7597e1a4b55aa3b4d6c5b029991d3b824b5bd331af8d4ab1ed687d/PyYAML-5.4.1.tar.gz#sha256=607774cbba28732bfa802b54baa7484215f530991055bb562efbed5b2f20a45e (from https://pypi.org/simple/pyyaml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, !=3.4.*, !=3.5.*). Command errored out with exit status 1: /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpvvbbpfl9 Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pyyaml==5.4.1 (from versions: 3.10, 3.11, 3.12, 3.13b1, 3.13rc1, 3.13, 4.2b1, 4.2b2, 4.2b4, 5.1b1, 5.1b3, 5.1b5, 5.1, 5.1.1, 5.1.2, 5.2b1, 5.2, 5.3b1, 5.3, 5.3.1, 5.4b1, 5.4b2, 5.4, 5.4.1)
ERROR: No matching distribution found for pyyaml==5.4.1_

Ik heb geprobeerd te begrijpen waar het fout gaat, maar dat lukt mij niet.
Wat kan ik doen om wel te kunnen upgraden laatste versie.

jossie67, 2021-04-26:
upgrade to python 3.7 heeft niet mogen baten

jossie67, 2021-04-26:
Ik ga eerst wel ff Ubuntu upgraden naar 20.04

closed 2021-04-26 by jossie67

dennissiemensma, 2021-04-26:
Je kunt recente meldingen van anderen bekijken en hun fix proberen. Grote kans dat een van die fixes het al voor je oplost: https://github.com/dsmrreader/dsmr-reader/issues?q=is%3Aissue+pyyaml+is%3Aclosed