dsmrreader/dsmr-reader #1375: MQTT-messages queuen indien MQTT-server niet beschikbaar?

dajappie, 2021-04-28:
Hi Dennis,

Soms valt de wifi-verbinding van mijn RPi weg, en het lijkt erop dat de messsages die naar de externe MQTT-server op de NAS gestuurd zouden moeten worden dan niet binnen DSMRreader gequeued worden. Daardoor valt een deel van de gelogde data weg. Ik zie zelf 2 opties:

- Een lokale MQTT-bridge draaien op de RPi zodat DSMRreader altijd kan afleveren, en de bridge levert dan af indien beschikbaar op de MQTT-server op de NAS. Mosquitto is niet by default reliable geconfigureerd, en zeker op een RPi met een SD-kaart kan dat flink wat I/O genereren om continue te loggen
- Misschien een manier om berichten in een bepaald timeframe opnieuw te genereren binnen DSMRreader bv. vanuit de admin pagina's?
- Zelf de data extraheren uit Postgres en via wat tooling richting MQTT uitsturen

Is hier iets slims voor te bedenken misschien?

dennissiemensma, 2021-04-28:
Bedankt voor je melding. Je zou kunnen experimenteren met de "Quality Of Service" instelling bij de Broker-instellingen in DSMR-reader. Heb je al geprobeerd om die op **QoS 2** te zetten? 
Standaard staat die namelijk op **QoS 0**, die "snel" is mar onbetrouwbaar. QoS 2 wacht op bevestiging van de broker en dat is eigenlijk wat je in jouw situatie wil.

Als dat niet werkt dan zal ik er een keer wat dieper naar kijken, want het lijkt me de bedoeling dat DSMR-reader bij QoS 2 pas het bericht in DSMR-reader zelf verwijdert als het verzenden gelukt is. Ik heb dat alleen zelf niet expliciet getest destijds.

dennissiemensma, 2021-04-28:
Hier staat overigens nog wat meer informatie over QoS en wanneer wat nodig is: https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/

Ik heb er zelf **geen** ervaring mee, maar ik zag dat de (Paho) MQTT-client in DSMR-reader er ondersteuning voor had dus ik heb dat ooit in #509 toegevoegd. 
Als het niet goed werkt dan kan ik er wel een keer wat dieper naar kijken, want ik weet niet precies wat de MQTT-client doet bij QoS 2 zonder verbinding. Ik zou verwachten dat die oneindig wacht tot er weer een verbinding is.

dajappie, 2021-04-28:
Hi Dennis, de QoS 2 instelling had ik inderdaad al gedaan op DSMRreader, dit omdat ik ditzelfde omgekeerd ook gebruik om MQTT weer uit te lezen vanuit Node-Red die het dan weer (o.a.) in InfluxDB gooit. QoS 2 zorgt dat het inderdaad altijd maar 1x wordt verstuurd, maar of Paho ook netjes queued als het niet lukt is dan de vraag. Zal anders eens kijken of ik direct vanuit DSMR-reader naar InfluxDB zal laten schrijven nu dat ook ondersteund wordt. Geen idee of die wel queued als het niet lukt? Fijne van MQTT/Node-Red ertussen is dat je wat meer kan spelen met de data voordat dit naar Influx gaat. If it ain't broken... Thanks!!

dennissiemensma, 2021-04-28:
Ik ga nog wel een keertje kijken hoe het precies zit met QoS 2 in Paho. Ik zag wel in de code dat die wacht op een bevestiging van de broker, maar misschien gaat dat dan toch niet altijd goed. 
Sowieso lost alleen QoS 2 het niet op bij langere downtime, omdat de queue in DSMR-reader momenteel max 200 berichten is, maar dat is nog wel vrij eenvoudig op te hogen of configureerbaar te maken.

dajappie, 2021-04-29:
200 berichten is inderdaad wel krap met 1 per 10 seconden, dan is een downtime van een half uurtje de max. Denk dat ik toch maar eens kijk voorlopig naar een lokale MQTT-bridge met persistence en een queue van een paar 1000 berichten. Dank!

dennissiemensma, 2021-04-29:
Die 200 was destijds een willekeurig gekozen waarde hoor en absoluut niet de limiet. Ik zal de default wat verhogen maar er ook voor zorgen dat het via een envvar instelbaar wordt.

Als je geen Docker draait, kun je dit toevoegen, onderaan `dsmrreader/settings.py`:
```
...

DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE = 5000
```
En **herstart** de applicatie dan.

----

Dat zeggende hebben lost dit niets op **als** DSMR-reader (of Paho) niet goed omgaat met QoS 2. Maar je kunt het zeker proberen.

Ik heb gisteren even kort getest, maar dat ging vrij moeizaam omdat elke keer wanneer ik de broker afknalde, DSMR-reader dat direct zag en de backend herstartte (en bleef herstarten). Het kan zijn dat een wegvallende wifi-verbinding echter iets anders werkt, dat kan ik niet goed simuleren.

dennissiemensma, 2021-04-29:
Voor de volgende release wordt `DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE` een nieuwe env var.

Ik zal ook nog kijken of ik wat meer kan debuggen met QoS 2 of dat uberhaupt iets toevoegt.

dennissiemensma, 2021-04-29:
Ik ben er nu achter dat ik voor QoS 2 nog wat meer moet doen in DSMR-reader, dus ik neem dat ook mee voor de volgende release. Daarmee zou de queue met QoS 2 betrouwbaar genoeg moeten zijn.

Het blijkt namelijk dat de huidige implementatie in DSMR-reader niet wacht op reactie van de broker. Bij QoS 0 is dat prima, maar bij 1 en 2 niet.

dennissiemensma, 2021-04-29:
Je kunt eventueel later de pre-release proberen om te zien of dat het issue definitief voor je oplost. Ik vermoed van wel, helemaal icm een grote queue.

dennissiemensma, 2021-05-10:
Het is nu onderdeel van [v4.16](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.16.0) dus probeer het vooral!

In jouw geval wil je waarschijnlijk de nieuwe envvar `DSMRREADER_MQTT_MAX_MESSAGES_IN_QUEUE` op 5000 of iets zetten.

dennissiemensma, 2021-05-11:
Er is nog een aanvulling en bugfix voor QoS 0 via [v4.16.1](https://github.com/dsmrreader/dsmr-reader/releases/tag/v4.16.1)

closed 2021-05-11 by dennissiemensma

dennissiemensma, 2021-05-12:
@dajappie ben je inmiddels in de gelegenheid geweest om het te proberen, gecombineerd met QoS 2 en eventueel een grotere queue?

dajappie, 2021-05-12:
Hi @dennissiemensma sorry nog niet ben op vakantie… Als ik terug ben zal ik een test doen. Overigens had ik wel al (ook) al een lokale MQTT-bridge obv Mosquitto neergezet met persistence en een queue van 10K berichten, dus er is ook al een fallback.

dajappie, 2021-06-14:
Hi @dennissiemensma, eindelijk even kunnen testen, lokale MQTT uitgezet en de queue in de admin pages loopt langzaam op. Heb het niet heel lang aan laten staan, maar het werkt zo te zien prima. Kan me voorstellen dat het mooi is als op de status-pagina een warning/error getoond wordt als de MQTT queue te groot wordt misschien? Normaliter zou er ik denk hooguit 0, 1 of hooguit enkele items gequeued moeten staan, daarna klopt er denk ik iets niet bij de meesten.

dennissiemensma, 2021-06-15:
@dajappie bedankt voor het proberen en fijn om te horen!

Er wordt momenteel een waarschuwing getoond als de queue vol zit, maar gezien de queue tegenwoordig stukken groter is, zal ik de drempel voor de waarschuwing wel wat lager leggen. Dat zeggende hebben helpt het uiteraard alleen als er ook actief gemonitored wordt op meldingen in DSMR-reader.