dsmrreader/dsmr-reader #1376: Meest recente meting update niet

KobeBergmans, 2021-04-29:
Ik heb nadat ik mijn ubuntu installatie had laten crashen een volledig nieuwe installatie gedaan van mijn besturingssysteem (nu Ubuntu server 20.04.02). Ik heb dan ook mijn dsmr-reader setup opnieuw geinstalleerd en de database opnieuw opgebouwd vanuit een backup. Alles lijkt nu terug vlot te werken alleen krijg ik op het dashbord niet de meest recente meting maar een van 1 maand en 1 week geleden (zie bijgevoegde schermafbeelding). De metingen komen wel door van mijn remote datalogger want de grafieken en overzichten zijn wel allemaal correct.

[image]

Setup:

- Ubuntu server 20.04.02
- Native install met remote datalogger
- Versie DSMR-reader v4.15.2

DSMR debug dump:
```

DSMR-READER
    App / Python / Database                                                     v4.15.2 / v3.8.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings                                             "None" / "101"

DATA
    Telegrams total (est.)                                                                              66808
    Consumption records electricity / gas (est.)                                                55674 / 24679

```

dennissiemensma, 2021-04-29:
Bedankt voor je melding. Staan er toevallig ook foutmeldingen in de "About" pagina of de debug-dump over dat de dagstatistieken achter lopen?

KobeBergmans, 2021-04-29:
Nee daar staat dat alles soepel lijkt te draaien.

Toen ik de database had geimporteerd dan had de applicatie enkele minuten nodig om alle statistieken terug te genereren maar ik zie al mijn statistieken waarvan de data in de backup zat

dennissiemensma, 2021-04-29:
Wellicht dat er dan puur visueel iets niet goed gaat. Wat geven deze twee queries voor output?

```
sudo su - postgres
psql dsmrreader

select id, timestamp from dsmr_datalogger_dsmrreading order by id desc limit 1;
select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 1;
```

dennissiemensma, 2021-04-29:
Want ik zie namelijk dat DSMR-reader momenteel kijkt naar het hoogste ID en niet perse het laatste telegram. Wellicht dat er iets bij de backup-restore niet lekker is gegaan qua ID's, waardoor nieuwe metingen een lager ID krijgen en dus niet gebruikt wordt. Maar dat is een aanname.

KobeBergmans, 2021-04-29:
De output van de 2 commando's
"""
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading order by id desc limit 1;
    id    |       timestamp
----------+------------------------
 30772191 | 2021-03-18 02:00:00+00
(1 row)
"""

"""
dsmrreader=# select id, timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 1;
  id   |       timestamp
-------+------------------------
 87992 | 2021-04-29 18:05:10+00
(1 row)
"""

Het lijkt mij ook puur visueel want de grafieken lijken wel te kloppen. En ja het lijkt inderdaad er iets mis is gelopen met de ID's want de datum met het hoogste id komt overeen met de datum die ik krijg bij de laatste meting die geprint wordt

dennissiemensma, 2021-04-29:
Ik zal het iig aanpassen voor een volgende release.

Los daarvan weet ik niet wat er gaat gebeuren als je nieuwe ID's de oude inhalen. Je kunt nog even dit proberen, die zou de "auto increment" moeten resetten:
```
sudo su - dsmr
 ./manage.py dsmr_sqlsequencereset
```

Mogelijk fixt dat ook indirect het visuele issue.

KobeBergmans, 2021-04-29:
Dat commando heeft ervoor gezorgd dat het nu terug in orde is, visueel en in de database krijg ik dezelfde metingen bij beide commando's.

Dankuwel!

dennissiemensma, 2021-04-29:
Top, hoe dan ook bedankt voor je melding. Voor de volgende release heb ik de sorting aangepast zodat dit ook zonder die sequence-reset goed gaat.

closed 2021-04-29 by dennissiemensma