dsmrreader/dsmr-reader #1378: Verbruikskosten niet meer zichtbaar 29-04-2021 (v4.15.2)

vrmware, 2021-04-30:
Goedemorgen Dennis,

Ik zie na 28 april geen verbruikskosten (costs veld dashboard is leeg). De costs bij de energie contracten zijn wel zichtbaar. Zowel in de web-interface als in het dagelijkse overzicht dat ik via Pushover op mijn telefoon ontvange zie ik geen verbruikskosten. 

DSMR-READER
    App / Python / Database                                                     v4.15.2 / v3.7.3 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 0.5s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                            2418190
    Consumption records electricity / gas (est.)                                                54277 / 21954

POSTGRESQL SIZE OF LARGEST TABLES (> 500 MB)
    public.dsmr_datalogger_dsmrreading                                                                 809 MB

```

dennissiemensma, 2021-04-30:
Bedankt voor je melding. Kan het zijn dat je huidige contract in dsmrreader toevallig de ingevulde einddatum heeft bereikt op de 28e?

vrmware, 2021-04-30:
Hoi Dennis,
Je hebt gelijk. Mijn contract is verlopen op 28 april. Ik ga het aanpassen. 
gr.
Vincent

vrmware, 2021-04-30:
Hoi Dennis,

Contract eind datum aangepast en alles werkt weer. 

gr.
Vincent

dennissiemensma, 2021-04-30:
Top, fijn om te horen!

closed 2021-04-30 by dennissiemensma