dsmrreader/dsmr-reader #1379: Grafieken worden elke 5 minuten opnieuw getekend

Element2, 2021-05-01:
DSMR-READER
    App / Python / Database                                                     v4.15.2 / v3.8.5 / postgresql
    BE sleep / DL sleep / Retention / Override                                     1.0s / 5.0s / 672h / False
    Latest telegram version read / Parser settings                                                 "50" / "4"

DATA
    Telegrams total (est.)                                                                              30403
    Consumption records electricity / gas (est.)                                                   2594 / 480

Raspberry PI 4B, 8GB-ram, 128GB-SD,

Alles werkt ok, zover ik kan zien geen fouten in de logs.
Wat opvalt is dat de grafieken in de 'live graphs' sectie precies elke 5 minuten opnieuw worden getekend.
Dus de oude grafieken verdwijnen, de spinner verschijnt kort en vervolgens worden alle grafieken opnieuw getekend.
Is dit 'normaal' gedrag?

dennissiemensma, 2021-05-01:
Bedankt voor je vraag. Dit is inderdaad "normaal", in de zin dat het een goedkope workaround is om sommige gegevens te verversen, die momenteel niet dynamisch worden opgehaald, zoals gasverbruik. Daarvoor is momenteel een paginarefresh nodig en daar is dit de workaround voor.

Ik kan het overigens uiteindelijk wel oplossen en alles dynamisch maken, echter kost zoiets enkele uren werk en ik vind het wat lastig te zeggen of dat het waard is. 
Daarnaast heeft DSMR-reader (bewust) geen tracking en ik heb daardoor dus ook geen statistieken over hoelang de meeste gebruikers die pagina gemiddeld open hebben of er uberhaupt (lang) naar kijken. Vandaar dat het meestal wat natte vinger werk is om te kiezen voor een goede oplossing en de bijbehorende inspanning.

Element2, 2021-05-01:
Ok, dank voor de uitleg. Het is niet heel hinderlijk en vroeg het me alleen af.
Overigens verders, complimenten voor het programma!

closed 2021-05-01 by Element2