dsmrreader/dsmr-reader #1380: Ondersteuning voor InfluxDB 2.0?

wderuijter, 2021-05-07:
@dennissiemensma wordt influxDB 2.0 ook ondersteund? 

Ik krijg een foutmelding wanneer ik ter test het script aanroep:
`./manage.py dsmr_influxdb_export_all_readings --to-influx-database export-test-2 --max-batches 100`

Namelijk een 401:
`raise InfluxDBClientError(err_msg, response.status_code)
influxdb.exceptions.InfluxDBClientError: 401: {"code":"unauthorized","message":"Unauthorized"}`

Zelfs wanneer ik de influxdb admin account credentials invul in de dsmr-reader config.

Ik zie dat er in dsmr-reader geen config veld is om een influxdb token op te slaan. Dus misschien ben ik wel iets aan het proberen wat (nog) niet ondersteund wordt in dsmr-reader :)

https://docs.influxdata.com/influxdb/v2.0/security/

dennissiemensma, 2021-05-07:
Bedankt voor je vraag. De integratie met Influx is vrij beperkt. Als ik het zo lees schijnt er de mogelijkheid te zijn om legacy auth in te schakelen.

Je zou dit kunnen proberen: 
- https://docs.influxdata.com/influxdb/v2.0/upgrade/v1-to-v2/manual-upgrade/#create-a-1x-compatible-authorization
- https://docs.influxdata.com/influxdb/v2.0/reference/cli/influx/v1/auth/

wderuijter, 2021-05-08:
@dennissiemensma bedankt voor je suggestie, ik heb de legacy auth ingeschakeld maar de rechten daarin gaan volgens mij niet verder dan read/write op een of meer buckets. De 401 is daarmee opgelost maar dan loop ik aan tegen:
`influxdb.exceptions.InfluxDBClientError: not implemented: CREATE DATABASE`.

dennissiemensma, 2021-05-08:
Je zou nog kunnen proberen om de database met de hand aan te maken. Wellicht is dat een eenmalige actie.

dennissiemensma, 2021-05-08:
Het lijkt er trouwens op dat de huidige InfluxDB client in DSMR-reader niet compatible is met 2.x. Er is wel een opvolger, maar die is weer niet compatible met versies ouder dan 1.8.

- Huidige: https://github.com/influxdata/influxdb-python
- Andere: https://github.com/influxdata/influxdb-client-python

Ik kan het daardoor niet zomaar opgraden. Dat zou met een volgende major release moeten. Deze staat sowieso al wel op de planning voor ergens in de komende maanden, maar daar heb je op dit moment dus weinig aan.

wderuijter, 2021-05-09:
Thanks voor je duidelijke toelichting @dennissiemensma

brainkiller, 2021-05-21:
Even volgen omdat ik hier ook gebruik van wil maken

dennissiemensma, 2021-11-17:
Ik zal kijken of ik nog forward-compattibility kan doen voor Influx 1.8, gezien dat erin zit, in de nieuwe influx 2.x Python client: 
- https://github.com/influxdata/influxdb-client-python#influxdb-18-api-compatibility
- https://github.com/influxdata/influxdb-client-python/blob/master/examples/influxdb_18_example.py

Dat zeggende hebben moet het niet te veel moeite kosten, want dan steek ik liever tijd in gewoon Influx 2.x config in DSMR-reader.

dennissiemensma, 2021-11-17:
Aardig eindje gekomen, maar ik denk dat het te complex is om zowel InfluxDB 1.8+ als 2.x te ondersteunen. Het was leuk geweest, maar te veel edge-cases en restricties. Plus dat ik dan in theorie weer een jaar of langer dat in stand moet houden voordat 1.x definitief weg kan.

_Het is namelijk nog steeds een experimentele/oppervlakkige integratie in DSMR-reader, dus ik denk dat het beter een **geforceerde** upgrade kan worden naar InfluxDB 2.x voor gebruikers. Scheelt tevens legacy en ik vermoed dat de meeste gebruikers van InfluxDB hun weg wel vinden in hoe te migreren naar een nieuwe Influx._
_Dit geldt overigens ook voor oude PostgreSQL/MySQL gebruikers met DSMR-reader, die moeten ook handmatig upgraden voordat ze naar DSMR-reader 5.x gaan._

dennissiemensma, 2021-11-21:
Klaargezet als onderdeel van v5 in 608cfef8ba0eea824d5272dea3eaa35a214a1e0f. 

Mag nog wel wat beter getest worden in de praktijk, al zou het in de basis niet veel anders moeten zijn. Goede reden om later voor v5 eerst een release candidate te doen.

closed 2021-11-21 by dennissiemensma

dennissiemensma, 2022-01-30:
DSMR-reader v5 is bijna klaar voor release. Daar zit dus ook deze feature in.

Echter gebruik ik zelf geen InfluxDB, dus is er iemand hier die op korte termijn de integratie wil testen/uitproberen? Dat betekent wel een vroegtijdige upgrade naar v5, maar die zal inhoudelijk toch niet meer veranderen. Hooguit op bugfixes na.

dennissiemensma, 2022-01-30:
Ik draai overigens uiteraard nu eerst zelf even v5, maar dat is meer vanuit sanity dat DSMR-reader uberhaupt blijft werken. InfluxDB is daar een relatief klein onderdeel van

webnar, 2022-01-30:
Ik wil het graag testen.

dennissiemensma, 2022-01-30:
Thanks! Ik kom later deze week bij je terug. Als je trouwens Docker draait, zal ik eerst met Xirixiz een docker image moeten maken bedenk ik me net.

webnar, 2022-01-30:
Ik draai idd docker. Laat maar weten wanneer die beschikbaar is om te testen.

dennissiemensma, 2022-01-30:
Voor nu alvast even een TODO gemaakt in https://github.com/xirixiz/dsmr-reader-docker/issues/264

webnar, 2022-02-02:
Ik draai momenteel de develop versie ik ga er even mee spelen en kijken of ik bijzonderheden tegen kom.

dennissiemensma, 2022-02-03:
Fijn, ben benieuwd!

dennissiemensma, 2022-02-04:
@webnar ben je inmiddels nog wat wijzer geworden of de integratie met InfluxDB 2.x goed genoeg is?

webnar, 2022-02-04:
Ik heb het nu enkele dagen draaien en die informatie komt correct in Influxdb. Nog in de logging van Influxdb of van damr-reader kom ik bijzonderheden tegen. Het ziet er dus erg goed uit. Mocht ik wat vreemds tegen komen dan zal ik het melden.

dennissiemensma, 2022-02-04:
Bedankt voor je snelle update! In dat geval vertrouw ik er wel op dat het werkt, want inhoudelijk zijn er verder ook geen wijzigingen geweest, puur de verbinding/authenticatie.

webnar, 2022-02-05:
De geschiedenis importeren werkt alleen niet. 

- Export draait zonder enige foutmelding.
- Bucket wordt gecreëerd
- Metingen (tabellen) worden gecreëerd

- Maar geen waardes zijn geïmporteerd

Draai het commando vanuit de container zoals vermeld:
 ./manage.py dsmr_influxdb_export_all_readings --to-influx-bucket dsmrreader-export

dennissiemensma, 2022-02-06:
Dank voor de terugkoppeling! Als het goed is werkt het juist alleen nog voor InfluxDB-2, maar ik heb puur het exporteren getest. 

Wat geeft `dsmr_influxdb_export_all_readings` aan tijden het draaien? Of zie je geen output?