dsmrreader/dsmr-reader #1381: Stuck at DSMR v4.10

ju1ced, 2021-05-07:
Had al even de updates niet opgevolgd en probeerde vandaag te updaten naar de laatste versie maar dat lijkt me niet te lukken.
als ik de ./deploy.sh run krijg ik volgende output

```
 --- You are currently running version:
4.10

 --- Checking Python version.
Running Python 3.8.5 (default, Jan 27 2021, 15:41:15)
[GCC 9.3.0]

 --- Pulling remote repository for new commits...

 --- Merging/updating checkout.
Updating 2c1ce855..2259101b
error: Your local changes to the following files would be overwritten by merge:
        dsmrreader/config/base.py
Please commit your changes or stash them before you merge.
Aborting

 >>> Running post-deployment script. <<<

 --- Checking whether VirtualEnv is activated.
     [i] ----- Activating 'dsmrreader' VirtualEnv...

 --- Checking Python version.
Running Python 3.8.5 (default, Jan 27 2021, 15:41:15)
[GCC 9.3.0]

 --- Checking & synchronizing base requirements for changes.
Requirement already up-to-date: autopep8==1.5.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 1)) (1.5.4)
Requirement already up-to-date: coreapi==2.3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.3.3)
Requirement already up-to-date: crcmod==1.7 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 3)) (1.7)
Requirement already up-to-date: django==3.1.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.1.4)
Requirement already up-to-date: django-admin-rangefilter==0.6.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 5)) (0.6.3)
Requirement already up-to-date: django-admin-sortable==2.2.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 6)) (2.2.3)
Requirement already up-to-date: django-colorfield==0.3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 7)) (0.3.2)
Requirement already up-to-date: django-debug-toolbar==3.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 8)) (3.2)
Requirement already up-to-date: django-filter==2.4.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 9)) (2.4.0)
Requirement already up-to-date: djangorestframework==3.12.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 10)) (3.12.2)
Requirement already up-to-date: django-solo==1.1.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 11)) (1.1.3)
Requirement already up-to-date: dropbox==10.10.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 12)) (10.10.0)
Requirement already up-to-date: gunicorn==20.0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 13)) (20.0.4)
Requirement already up-to-date: influxdb==5.3.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 14)) (5.3.1)
Requirement already up-to-date: paho-mqtt==1.5.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 15)) (1.5.1)
Requirement already up-to-date: pyserial==3.4 in /usr/lib/python3/dist-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 16)) (3.4)
Requirement already up-to-date: pyserial-asyncio==0.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 17)) (0.4)
Requirement already up-to-date: python-dateutil==2.8.1 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 18)) (2.8.1)
Requirement already up-to-date: python-decouple==3.3 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 19)) (3.3)
Requirement already up-to-date: pytz==2020.4 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 20)) (2020.4)
Requirement already up-to-date: pyyaml==5.3.1 in /usr/lib/python3/dist-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 21)) (5.3.1)
Requirement already up-to-date: requests==2.25.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 22)) (2.25.0)
Requirement already up-to-date: urllib3==1.26.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from -r dsmrreader/provisioning/requirements/base.txt (line 23)) (1.26.2)
Requirement already satisfied, skipping upgrade: toml in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from autopep8==1.5.4->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (0.10.2)
Requirement already satisfied, skipping upgrade: pycodestyle>=2.6.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from autopep8==1.5.4->-r dsmrreader/provisioning/requirements/base.txt (line 1)) (2.6.0)
Requirement already satisfied, skipping upgrade: coreschema in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (0.0.4)
Requirement already satisfied, skipping upgrade: itypes in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.2.0)
Requirement already satisfied, skipping upgrade: uritemplate in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (3.0.1)
Requirement already satisfied, skipping upgrade: asgiref<4,>=3.2.10 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from django==3.1.4->-r dsmrreader/provisioning/requirements/base.txt (line 4)) (3.3.1)
Requirement already satisfied, skipping upgrade: sqlparse>=0.2.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from django==3.1.4->-r dsmrreader/provisioning/requirements/base.txt (line 4)) (0.4.1)
Requirement already satisfied, skipping upgrade: six>=1.12.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from dropbox==10.10.0->-r dsmrreader/provisioning/requirements/base.txt (line 12)) (1.14.0)
Requirement already satisfied, skipping upgrade: setuptools>=3.0 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from gunicorn==20.0.4->-r dsmrreader/provisioning/requirements/base.txt (line 13)) (44.0.0)
Requirement already satisfied, skipping upgrade: msgpack in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from influxdb==5.3.1->-r dsmrreader/provisioning/requirements/base.txt (line 14)) (0.6.2)
Requirement already satisfied, skipping upgrade: idna<3,>=2.5 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.25.0->-r dsmrreader/provisioning/requirements/base.txt (line 22)) (2.8)
Requirement already satisfied, skipping upgrade: certifi>=2017.4.17 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.25.0->-r dsmrreader/provisioning/requirements/base.txt (line 22)) (2019.11.28)
Requirement already satisfied, skipping upgrade: chardet<4,>=3.0.2 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.8/site-packages (from requests==2.25.0->-r dsmrreader/provisioning/requirements/base.txt (line 22)) (3.0.4)
Requirement already satisfied, skipping upgrade: jinja2 in /usr/lib/python3/dist-packages (from coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (2.10.1)

 --- Applying database migrations.
Operations to perform:
[... 3 lines cut ...]

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 561 unmodified.

[... cut ...]
 * Reloading process: dsmr_datalogger                [!!] PID file does not exist or the datalogger is disabled (Should it run? As sudo user, try 'sudo supervisorctl start dsmr_datalogger')

 --- Clearing cache...

 --- Deployed version:
4.10

 >>> Deployment complete. <<<
```

de versie blijft op 4.10 en in de webinterface blijf ik update beschikbaar zien.
DSMR draait op een ubuntu 20.04 en de remote datalogger draait op een rpi2.
alles werkt wel nog steeds goed, maar ik kan dus niet updaten, enig idee wat dit kan zijn?

bedankt
Joost

dennissiemensma, 2021-05-07:
Bedankt voor je melding. De oorzaak lijkt dit:

```
 --- Merging/updating checkout.
Updating 2c1ce855..2259101b
error: Your local changes to the following files would be overwritten by merge:
        dsmrreader/config/base.py
Please commit your changes or stash them before you merge.
Aborting
```

Je hebt waarschijnlijk ooit wijzigingen gedaan aan `dsmrreader/config/base.py`. Om te voorkomen dat updaten voor merge-conflicten zorgt, weigert DSMR-reader te updaten.

Je kunt kijken wat de wijzigingen zijn en de evt bewaren en daarna resetten en alsnog upgraden:
```
sudo su - dsmr
git diff
```

Als je wijzigingen weg kunnen
```
git reset --hard HEAD
./deploy.sh
```

ju1ced, 2021-05-07:
dat was het inderdaad, bedankt voor je snelle reactie!