dsmrreader/dsmr-reader #1385: Tijdverschil

donnie-dominguez, 2021-05-12:
Toevallig viel me iets op vandaag. Als ik naar mijn gas-gebruik kijk in de live-grafiek dan is deze een uur versprongen met de grafiek in het 'Archive'. Een zomer/wintertijd dingetje?

Live grafiek
[image]

Archive
[image]

Ik gebruik DSMR-reader v4.15.2 op Ubuntu

dennissiemensma, 2021-05-12:
Bedankt voor je melding! Ik zal dit verder onderzoeken. Het is mij nooit opgevallen en het is even de vraag welke van de twee dan klopt.

donnie-dominguez, 2021-05-13:
Inmiddels geüpgraded naar DSMR-reader v4.16.2, maar daarin treed het ook op. Ik heb het gasverbruik even in de gaten gehouden, en ik denk te kunnen concluderen dat het 'probleem' in het archief zit.

Gister is er tussen 20.00 en 21.00 iemand onder de douche geweest hier in huis, de kolom gasverbruik waar de cursor op staat in onderstaande afbeeldingen geeft die weer. Vervolgens is er na 23.00 weer gedouched, wat in het archief weer is gegeven als 22.00

Live view
[image]

Archief
[image]

dennissiemensma, 2021-05-13:
Bedankt voor je aanvulling. Ik vermoed overigens dat dit iets kan zijn wat wellicht al hele lange tijd in DSMR-reader zit, maar niet eerder is opgevallen. Ook omdat ik de basisfunctionaliteit doorgaans zelden aanpas, maar het kan natuurlijk alsnog iets recents geweest zijn.

dennissiemensma, 2021-05-13:
Het eerste verschil dat ik bij mezelf kan vinden is vanaf 29 oktober 2019, dus nog los van de oorzaak lijkt het er al een tijdje in te zitten.

dennissiemensma, 2021-05-13:
Het zou te maken kunnen hebben met 4bdb7c378725fa85558764cbd94ec4240a7f4a8b / #766. Ik zie nog niet direct de relatie met de bug zelf, het is vast iets subtiels.

dennissiemensma, 2021-05-13:
De oorzaak lijkt uiteindelijk inderdaad te zitten in die wijziging van toen: https://github.com/dsmrreader/dsmr-reader/pull/766/files#diff-79b655a9839c8d926e6778808d4f0ffe7a31f9d6038143b972023d3dd9f5c56dR185

Ik verwijt mezelf vooral dat het ontbreekt aan goede automatische tests die dit moeten borgen, dus ik zal daar eea aan uitbreiden en het fixen. Ik weet echter niet of ik het ook met terugwerkende kracht kan en wil fixen, omdat niet iedereen alle oude data nog heeft en het is verder alleen visueel.
 De dagtotalen zouden namelijk wel moeten kloppen en die zijn het belangrijkste. Die pakken de eerste en laatste meting van de dag en gebruiken niet deze uurtotalen.

dennissiemensma, 2021-09-18:
Ik ben hiervoor al een tijdje bezig. De beste oplossing is wat rework en versimpeling van hoe statistieken nu gegenereerd worden. Het was eerst het doel dit voor de volgende release te doen, maar het raakt zoveel dat het beter in een aparte release erna kan.

dennissiemensma, 2021-10-18:
Ik heb een deel van 2ff635033df54089c89b8bd7a257f7480abe5ef0 teruggedraaid in de ontwikkelbranch en zal die een tijdje zelf draaien om te zien of dat het euvel oplost.

dennissiemensma, 2021-10-19:
De fix lijkt inderdaad te werken. Verschil is goed te zien tussen fouttief (zondag) en goed (gisteren).

## Overzicht
[image]

## Zondag
Uur verschil
[image]

## Gisteren
3x exacte match met grafiek qua tijd en waarde
[image]

dennissiemensma, 2021-10-19:
Wordt onderdeel van v4.19. Ik draai het nog even een paar dagen voor de zekerheid.

closed 2021-10-19 by dennissiemensma