dsmrreader/dsmr-reader #1386: Herinstallatie packages

reboot1983, 2021-05-12:
Onderstaande viel mij vanmorgen toevallig op. Het lijkt erop alsof het ./deploy.sh script niet (correct) de huidige versies detecteert. Het verwijderd de betreffende installatie van deze twee packages om vervolgens dezelfde versie weer te installeren.

[image]

Daarnaast een vraag over de outdated PIP melding. Is dit iets wat we veilig kunnen uitvoeren? Ik vind er niets over in de handleiding en weet ook niet of deze melding door DSMR-Reader wordt gegeven of gewoon een standaard melding is?

dennissiemensma, 2021-05-12:
Bedankt voor je melding! Ik zie echter niet helemaal wat je bedoelt. Dropbox 11.7 wordt weggehaald en vervangen door 11.8 onderaan. Idem voor Auto-pep 1.5.6, naar 1.5.7. Bedoel je dat?

Pip is zelf wat speciaal en niet perse onderdeel van DSMR-reader. Je kunt die echter gerust updaten met:
```
sudo su - dsmr
pip install --upgrade pip
```

closed 2021-05-13 by dennissiemensma

reboot1983, 2021-05-13:
Wauw.. Sorry daarvoor. Ik had beter moeten/mogen kijken.