dsmrreader/dsmr-reader #1387: Telegrammen worden niet verwerkt

vincedh, 2021-05-12:
Hoi Dennis,
Ik heb sinds vanmorgen versie v4.16.1 geïnstalleerd, maar nu worden nieuwe telegrammen niet verwerkt (Te veel onverwerkte telegrammen: 245   (4 minuten geleden)). Als ik m'n VM herstart worden de telegrammen verwerkt die er nog stonden, maar dan loopt de teller langzaam weer op met nieuwe telegrammen. Enig idee?
Hoor het graag weer! Bedankt vast.

vincedh, 2021-05-12:
Ik heb uitgevoerd:

sudo supervisorctl restart all

Nu lijkt het beter te gaan, ook nieuwe lijken nu verwerkt te worden. Mocht ik nog wat tegenkomen hoor je dat uiteraard.

vincedh, 2021-05-12:
Iets te vroeg gejuicht helaas, Te veel onverwerkte telegrammen: 9181   (2 uur geleden)
Als jij nog een idee hebt hoor ik het graag.

dennissiemensma, 2021-05-12:
Bedankt voor je vraag. Als je zulke issues hebt kun je denk ik het beste beginnen bij de logfiles:
- https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/logfiles.html
In dit geval heb je de log van dsmr_backend nodig.

Als daar niets in staat, kun je het beste tijdelijk debug-logging inschakelen (en herstarten) om te zien wat er precies gebeurt. Bovenstaande logfile zal dan veel meer informatie bevatten. 

Howto:
- https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html

vincedh, 2021-05-12:
Ik vermoed dat het iets met m'n MQTT verbinding te maken heeft en ik zag in je changelog dat daar eea is aangepast.

2021-05-12 19:51:09,824 | DEBUG    schedule.reschedule:72 | SP: Rescheduled "Generate consumption data" to 2021-05-12 19:51:24.823084+02:00 (ETA 0:00:14.999015)
2021-05-12 19:51:09,826 | DEBUG    persistent_clients.run:30 | CLIENTS: Running 1 active client(s)
2021-05-12 19:51:09,827 | INFO     broker.run:71 | MQTT: Processing 1 message(s) using QoS level 0
2021-05-12 19:51:09,827 | DEBUG    broker.run:74 | MQTT: Publishing queued message (#24779516) for dsmr/consumption/gas/currently_delivered: 0.000
2021-05-12 19:51:09,827 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending PUBLISH (d0, q0, r1, m7), 'b'dsmr/consumption/gas/currently_delivered'', ... (5 bytes)
2021-05-12 19:51:09,828 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) failed to receive on socket: [Errno 32] Broken pipe
2021-05-12 19:51:09,828 | DEBUG    broker.on_disconnect:140 | MQTT: (Paho on_disconnect) 1
2021-05-12 19:51:09,828 | WARNING  broker.on_disconnect:143 | MQTT: --- Unexpected disconnect, re-connecting...
2021-05-12 19:51:09,829 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b'slimmemeter'
2021-05-12 19:51:09,829 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:09,829 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:09,829 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:09,829 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:09,852 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Received CONNACK (0, 0)
2021-05-12 19:51:09,852 | DEBUG    broker.on_connect:124 | MQTT: (Paho on_connect) {'session present': 0} | 0
2021-05-12 19:51:09,852 | DEBUG    broker.on_connect:127 | MQTT: --- 10.0.42.3 : 1883 -> Connection successful
2021-05-12 19:51:09,852 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:29,873 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:51:49,879 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:52:09,898 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending PINGREQ
2021-05-12 19:52:09,898 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:52:09,898 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:52:09,901 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Received PINGRESP
2021-05-12 19:52:09,901 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:52:29,914 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:52:49,926 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:53:09,947 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending PINGREQ
2021-05-12 19:53:09,947 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:53:09,947 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:53:09,950 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Received PINGRESP
2021-05-12 19:53:09,950 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:53:29,969 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:53:49,990 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:54:10,010 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending PINGREQ
2021-05-12 19:54:10,010 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:54:10,011 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker
2021-05-12 19:54:10,013 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Received PINGRESP
2021-05-12 19:54:10,013 | DEBUG    broker.run:93 | MQTT: Waiting for message (#24779516) to be marked published by broker

dennissiemensma, 2021-05-12:
Het kan zijn dat er iets is met je verbinding naar de MQTT-broker. Sinds v4.16.1 handhaaft DSMR-reader striktere controle in de uitwisseling van MQTT-berichten. Ik zie in je log telkens hetzelfde bericht `24779516` voorbijkomen, die niet afgeleverd wordt.

En ook deze: `failed to receive on socket: [Errno 32] Broken pipe`, wat *kan* duiden op iets met een verbinding.

Het kan zijn dat de MQTT-timeout wat te hoog staat in DSRM-reader, maar dat zou alleen een issue moeten zijn als de verbinding slecht is (of de broker traag). Dit verklaart in ieder geval wel waarom het backend-proces niet toekomt aan andere zaken, alleen kan het zijn dat dit eerder ook al speelde.

Enig idee of je in oudere release MQTT-berichten miste? Het kan natuurlijk ook steeds een bug zijn in DSMR-reader, helemaal omdat het MQTT-gedeelte aangepast. Echter heb ik gisteravond wel wat varianten geprobeerd en dat *leek* toen goed. Voor zowel QoS 0, 1 als 2.

Je kunt nog proberen om in de MQTT-instellingen in DSMR-reader QoS level `2` te gebruiken. Nu lijkt die bij jou op `0` te zijn, ik ben benieuwd of dat verschil maakt.

dennissiemensma, 2021-05-12:
Het kan ook nog een bug in de MQTT-client in DSMR-reader (of hoe DSMR-reader het gebruikt) zijn. Want bovenin staan deze twee regels:
```
2021-05-12 19:51:09,827 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) Sending PUBLISH (d0, q0, r1, m7), 'b'dsmr/consumption/gas/currently_delivered'', ... (5 bytes)
2021-05-12 19:51:09,828 | DEBUG    broker.on_log:154 | MQTT: (Paho on_log) failed to receive on socket: [Errno 32] Broken pipe
```

Dus het publiceren mislukt, daaronder verbindt die opnieuw, maar blijft het hangen in niemandsland.

Hoe dan ook kun je sowieso dus QoS 2 proberen.

vincedh, 2021-05-12:
Ik durf niet te zeggen of ik voorheen MQTT berichten miste, maar ik zag in m'n broker wel regelmatig berichten voorbij komen, de verbinding werkte dus wel goed.
Ik heb DSMR nu op QoS 2 staan en voorlopig lijkt dat nu te werken...
Ik hou je op de hoogte.

vincedh, 2021-05-13:
Hoi Dennis,
Ook na een nachtje draaien blijven de telegrammen verwerkt worden. Het lijkt er dus op dat QoS op `2` zetten heeft geholpen. 
Wat mij betreft opgelost dus, als ik nog wat kan bekijken voor je hoe dit komt hoor ik het wel, anders laten we het hierbij.
Uiteraard weer hartelijk dank voor de support en een fijne Hemelvaartsdag.

dennissiemensma, 2021-05-13:
Bedankt voor je update en fijn om te horen! Ik zou het dan voor nu idd zo laten. 

Als er later toch nog wijzigingen nodig blijken in DSMR-reader, dan komt dat vanzelf weer. Ik wacht ook even af of nog meer MQTT-gebruikers hier tegenaan lopen.

closed 2021-05-13 by dennissiemensma