dsmrreader/dsmr-reader #1388: upgrade error...

newtech2015, 2021-05-12:
hallo,

ik gebruik nu dsmr v.4.6, maar wilde nog eens upgraden naar de laatste nieuwe versie.
zowel bij 4.16.2 als bij 4.16.1 eindigt de update in een error:

ERROR: Exception:
Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/cli/base_command.py", line 216, in _main
    status = self.run(options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/cli/req_command.py", line 182, in wrapper
    return func(self, options, args)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/commands/install.py", line 325, in run
    reqs, check_supported_wheels=not options.target_dir
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/resolution/legacy/resolver.py", line 183, in resolve
    discovered_reqs.extend(self._resolve_one(requirement_set, req))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/resolution/legacy/resolver.py", line 388, in _resolve_one
    abstract_dist = self._get_abstract_dist_for(req_to_install)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/resolution/legacy/resolver.py", line 340, in _get_abstract_dist_for
    abstract_dist = self.preparer.prepare_linked_requirement(req)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/operations/prepare.py", line 483, in prepare_linked_requirement
    req, self.req_tracker, self.finder, self.build_isolation,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/operations/prepare.py", line 91, in _get_prepared_distribution
    abstract_dist.prepare_distribution_metadata(finder, build_isolation)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/distributions/sdist.py", line 38, in prepare_distribution_metadata
    self._setup_isolation(finder)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_internal/distributions/sdist.py", line 96, in _setup_isolation
    reqs = backend.get_requires_for_build_wheel()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/wrappers.py", line 161, in get_requires_for_build_wheel
    'config_settings': config_settings
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/wrappers.py", line 265, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pip._vendor.pep517.wrappers.BackendUnavailable: Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/pip/_vendor/pep517/_in_process.py", line 86, in _build_backend
    obj = import_module(mod_path)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 941, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/local/lib/python3.6/site-packages/setuptools/__init__.py", line 6, in <module>
    import distutils.core
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/site-packages/_distutils_hack/__init__.py", line 83, in create_module
    return importlib.import_module('._distutils', 'setuptools')
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'setuptools._distutils'

wat gaat er mis, en wat kan ik eraan doen?

dennissiemensma, 2021-05-13:
Bedankt voor je melding. Je bent niet de eerste, je zou dit kunnen proberen: #1307 / #1334

newtech2015, 2021-05-13:
dennis,

bedankt, heb nu weer de laatste nw versie draaien!

dennissiemensma, 2021-05-13:
Top, fijn om te horen!

closed 2021-05-13 by dennissiemensma