dsmrreader/dsmr-reader #1390: Pie charts in trends-pagina vervangen door bar/line chart

johannienhuis, 2021-05-15:
De pi-charts op de trends-pagina zijn onoverzichtelijk. Daarbij zijn de verschillende uren moeilijk onderling vergelijkbaar (vooral voor de delen van een pie die niet naast elkaar liggen).

Is het mogelijk om de pie chart te vervangen door een normale grafiek zoals op de archive-pagina getoond wordt? Hierdoor zijn alle uren van een dag in 1 oogopslag goed vergelijkbaar. Bovendien kan de gebruiker dan eenvoudig tussen bar/line chart kiezen.

dennissiemensma, 2021-05-16:
Bedankt voor je vraag. Zou je een screenshot willen maken van je eigen situatie? Dit diagram is momenteel druk, dus het is sowieso een uitdaging om hem in alle sitauties leesbaar te maken. 
Het is momenteel bewust een variant op een pie-chart, die de onderlinge verschillen juist wat meer zou moeten benadrukken. Ik zou een keer kunnen kijken voor een tweede variant, echter wil ik liever zo min mogelijk variaties en het liefste maar 1 of 2 die voor iedereen leesbaar zijn. Ongeacht in welke vorm dat dan uiteindelijk is.

johannienhuis, 2021-05-16:
Dank voor je reactie. Zie bijgaand voorbeeld zoals deze bij mij zichtbaar is. [Trends  DSMR Reader.pdf](https://github.com/dsmrreader/dsmr-reader/files/6489288/Trends.DSMR.Reader.pdf)

johannienhuis, 2021-05-16:
Overigens ben ik zelf niet zo'n voorstander van het gebruik van Pie charts. Veel data visualisatie guru's adviseren om deze niet te gebruiken simpelweg omdat er bijna altijd betere alternatieven zijn. (Google maar eens op 'stephen few pie charts'). Ken je https://chart.guide/? Ook hier zie je dat de pie chart afgeraden wordt.

En zeg nou zelf, dit diagram is veel overzichtelijker :-)
[image]
(afkomstig uit de archive-pagina / dag-niveau)

dennissiemensma, 2021-05-17:
Bedankt voor je aanvullingen. Wat betreft je eigen screenshots snap ik goed wat je bedoelt. De eerste pie is wel zoals het bedoeld is en is om duidelijk aan te geven welk uur of uren dominant zijn. 
Echter, de situatie in je tweede en derde pie heb ik zelf nog niet eerder gezien en dat maakt het inderdaad niet duidelijk. Ter vergelijking, ik heb wel behoorlijk wat rare situaties gehad en de meeste kunnen fixen met een een workaround, maar zo heb ik het meestal gezien:

[image]
En hoewel je natuurlijk nog kan discussiëren over of dat de ideale weergave is, vond ik het wel werkbaar.

Ik heb ook nog even gezocht naar je tips over de pie's, maar dat gaat vermoedelijk vooral op voor normale pie's met vele stukken, zoals [deze](https://echarts.apache.org/examples/en/editor.html?c=pie-legend). Dat was ook de reden dat ik bewust gekozen had voor de [rose-variant](https://echarts.apache.org/examples/en/editor.html?c=pie-roseType-simple), die schaalt naar mate van de dominantie.

Verder heb ik ook [deze variant](https://echarts.apache.org/examples/en/editor.html?c=scatter-single-axis) nog overwogen, maar die is denk ik alleen duidelijk als het echt veel datapunten heeft, zoals die in het voorbeeld. 
Ik zal iig voor je kijken of een staafdiagram het duidelijker maakt, wellicht evt gesorteerd van hoog naar laag zoals [deze](https://echarts.apache.org/examples/en/editor.html?c=bar-race) (zonder animatie dan). En eventueel later nog samengevoegd met een andere variant oid in deze [opties](https://echarts.apache.org/examples/en/index.html#chart-type-bar).

johannienhuis, 2021-05-18:
Bedankt voor je inhoudelijke reactie! Hierbij weer een aanvulling van mijn kant.

Het voorbeeld dat jij meestuurt ziet er qua structuur wel beter uit. Maar ik blijf een taart diagram nadelig vinden omdat de oppervlakte-maat in een circel moeilijker te vergelijken is dan de hoogte-maat van staafdiagram. 

Als je een chart wil gebruiken waarbij je oppervlakte als maatvoering gebruikt dan vind ik persoonlijk alleen vierkante vormen geschikt. (Bijvoorbeeld Treemap of een Marimekko chart maken ook gebruik van oppervlakte als visualisatie-hulp. Treemap en Marimekko vind je hier: https://chart.guide/charts/part-to-whole-or-composition/ (in de 3e rij). Dit even ter illustratie hoe je (naar mijn mening) met oppervlakte kunt werken in charts echter deze charts zijn denk ik niet inzetbaar voor de trends-pagina).

Je hebt in je reactie ook nog een hele interessante aanvulling: https://echarts.apache.org/examples/en/editor.html?c=scatter-single-axis. Wat ik hier echt super interessant aan vind is de onderverdeling naar dagen en uren zodat je beter zicht krijgt op wekelijkse patronen. Alleen de oppervlakte vergelijking in circels vind ik dus minder sterk in dit voorbeeld... ;-) 

Zou je als alternatief op de dagen/uren visualisatie de horizon-chart eens willen overwegen? (Zie rechtsonder in deze link https://chart.guide/charts/time-or-trend/). Dit zou denk ik een heel sterk alternatief zijn, vraag is natuurlijk altijd of dit te realiseren is ;-)

Nogmaals dank voor je reactie. Ik denk graag met je mee. Zoals je merkt heb ik een eigen mening over data visualisaties maar voel je vrij om daar wel of niet iets mee te doen.

dennissiemensma, 2022-07-06:
Ik heb nog deze variant geprobeerd, maar rekt te veel uit:

[image]

---

Vanaf DSMR_reader v5.4 wordt het zoiets:

[image]

De onderste pie heb ik zoo gehouden, omdat het prima verhouding geeft tussen twee (doorgaans relatief gelijk aan elkaar) vlakken

closed 2022-07-06 by dennissiemensma