dsmrreader/dsmr-reader #1391: Datalogger stopt met het verwerken

mirkin1993, 2021-05-16:
Sinds de laatste paar dagen stopt datalogger met het verwerken van de uitgelezen 'dsmr-uitlezing'. Na een reboot werkt het weer naar behoren. 
Waar kan dit aan liggen?

dennissiemensma, 2021-05-16:
Bedankt voor je melding. Welke versie draai je? En anders zou je sws de stappen uit #1387 kunnen proberen. De logs zijn de eerste plek om te kijken.

dennissiemensma, 2021-05-17:
@mirkin1993 ben je inmiddels wat wijzer geworden uit de logs?

mirkin1993, 2021-05-17:
@dennissiemensma ik heb er net naar gekeken. Kennelijk was er een probleem met de MQTT broker, nadat ik de QoS op 2 heb gezet was het probleem voorbij.

dennissiemensma, 2021-05-17:
Bedankt voor je update! Draaide je al wel de laatste versie van DSMR-reader? 

Zo ja dan ben ik wel benieuwd naar de logs en of je dezelfde `PINGREQ` / `PINGRESP` ziet als diegene in het andere issue. Zo ja dan is het simpelweg nog een bug in DSMR-reader, want hoewel QoS 2 aan te raden is, moet QoS 0 ook gewoon goed (gaan) werken.

mirkin1993, 2021-05-17:
@dennissiemensma Ik draai: 4.16.2
Dit zijn de laatste regels van mij log of heb je meer nodig? En het stukje `PINGREQ` / `PINGRESP` moet je mij denk ik even uitleggen.
```
2021-05-13 07:34:59,458 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client no longer connected
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
2021-05-13 07:39:06,953 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client no longer connected
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/v4/troubleshooting.html#logging
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-13 18:48:09,528 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-13 18:53:10,436 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-13 18:56:05,797 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-13 19:00:04,044 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-14 07:10:16,231 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 07:05:12,155 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 07:11:01,029 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 07:29:14,805 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 07:58:03,700 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 08:02:01,542 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 08:10:41,819 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 08:16:06,296 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-15 08:20:03,911 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-16 06:19:07,619 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-16 06:26:29,721 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...
[... cut ...]
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
Current logging level set to "ERROR". More information can be found here: https://dsmr-reader.readthedocs.io/en/latest/how-to/troubleshooting/enabling-debug-logging.html
2021-05-17 20:41:15,643 | ERROR    persistent_clients.run:37 | CLIENTS: Run error: MQTT: Client loop() failed, requesting restart...

```

dennissiemensma, 2021-05-17:
Bedankt voor de details, het lijkt erop dat je nog een andere fout krijgt dan het andere issue, maar de kern is denk ik hetzelfde. Ik ga verder graven en zal het via #1393 proberen te fixen.

Voor nu kun je denk ik het beste QoS 2 blijven gebruiken (die dan ook het meest betrouwbaar) en mocht dat toch niet werken dan hoor ik het tzt weer graag!

closed 2021-05-17 by dennissiemensma