dsmrreader/dsmr-reader #1392: Toevoegen total costs huidg contract MQTT

hosmancevic, 2021-05-17:
Nu is er al het MQTT topic 'Current month/year totals', is het mogelijk om ook current energy contract totals te krijgen?
Op die manier kan ik dus zien hoeveel ik deze contract periode heb verbuikt ipv het kalenderjaar.

dennissiemensma, 2021-05-17:
Bedankt voor je verzoek. Ik zal het overwegen voor een toekomstige release. Heb je het puur voor de weergave nodig of sla je het ook nog ergens op? 

Als "buitenstaander" vind ik het wat lastig om te oordelen wat de beste oplossing is waar meeste gebruikers mee geholpen zijn. En of MQTT puur dan hiervoor gebruikt wordt omdat dat daadwerkelijk het beste werkt of omdat alternatieven (zoals een API) onbekend zijn.

Dit soort gegevens, net zoals de huidige maand/jaar-totalen, raken namelijk wel redelijk de grens van MQTT-complexiteit tov bijvoorbeeld gewoon het "on-demand" opvragen via de API van DSMR-reader. Helemaal omdat deze gegevens doorgaans ook niet meer verder geaggregeerd worden en puur ter weergave "as-is".

hosmancevic, 2021-05-17:
Top! Ja eigenlijk gebruik ik MQTT om de waardes naar HASSIO te sturen, en daar wordt het opgeslagen en draaien er ook automations om me te waarschuwen als ik meer verbruik dan geschat etc.

dennissiemensma, 2022-06-17:
Toevallig recent behandeld in #1639. Uiteindelijk niet gemaakt en via de API in #1640 een generiekere manier ontsloten.

closed 2022-06-17 by dennissiemensma